probonopd/MiniDexed #168: Serial MIDI (DIN->Optocoupler->GPIO15) issue

arsamus, 2022-04-25:
Hi everybody, i've build a MIDI IN interfaz via optocoupler to connect a keyboard vía tradicional midi cable (schematica below) using GPIO15 (UART RXD). Everything works ok but only if I press one key at a time. If I play chords, just one note is played by Mididexed and if a "note Off" event occurs near another event (in the same message chain) the note continues to playing forever. 

It seems that MIDI  messages chains with more than 3 bytes, for example when I play two notes at the same time or when pitch bender is actioned, are truncated and only the first 3 bytes are proceced and the rest of the message is los, as I could note on screen activating Midi dumping option.

I've tested on Raspberry 1B+ and 2B with the following schematics:
[image]
(schematics from [Here](https://forum.arduino.cc/t/receive-midi-in-arduino/373302) but I've adapted to RB with 3.3v)

craigyjp, 2022-04-25:
I don't have much success with 6n devices, I tend to use 4n devices, this one works perfectly with a midi keybed playing chords and sequences.[image]

craigyjp, 2022-04-25:
Here is the 6n138 version that I started with, check the wiring against yours.

[image]