probonopd/MiniDexed #173: Yamaha KX-49/KX-25 keyboards are recognized, but no data in the midi debug monitor

StephanBeer, 2022-04-25:
I have been building and trying MiniDexed the last 6 days on Raspberry Pi Zero, Raspberry Pi 1B and Raspberry Pi 3B, but on neither of them i got sound out of MiniDexed when connecting my Yamaha KXs ( https://synthpedia.net/yamaha/kx49/ ) via USB. First i thought i messed up the configuration files or its sending on the wrong midi channel, but after trying out all different kind of combinations and switching on MIDIDumpEnabled=1, i discovered that other midi controller from my music gear arsenal are sending data wich is received by minidexed, but not the KX keyboards.

I tried :
- Fishman Triple Play Guitar Midi Controller which is working fine, even ProgrammChanges
- Presonus Atom, a drumpad controller like Maschine...which almost fried the Raspberry as the controller was sucking so much power that i pulled the usb plug after some seconds....but MiniDexed received midi data and was producing sounds.

Like the other twos, the Yamaha KX keyboards are recognized and configured as you can see on the screenshots...but no data when being played.

Now heres the thing: The KX keyboards are presenting 3 midi ports when used in windows, MacOS or even in Linux which are YAMAHA KX MIDI 1, .. MIDI 2 and ...MIDI 3 (screenshot is from Dexed runing on ArchLinux, working perfectly fine there).
- MIDI 1 is for all the keys and the 4 encoder, which are doubled...so 8 of them, sending CCs
- MIDI 2 is for the transport section of the keyboard, which is for DAW usage only...MCU
- MIDI 3 is exclusively for a external Yamaha patch templates exchange software...not used anymore

My guess is that MiniDexed is messing up the 3 MidiPorts when doing the configuration and is listening on the wrong port , maybe Port 3 where theren aren´t any midi notes send out from playing the keyboard ??

"I am owner of a Yamaha KX-88 (the old tank from 1985 https://yamahablackboxes.com/collection/yamaha-kx88-controller/ ) for 12 years now, but sadly the master midi keyboard collected some dust over the last years. Would love to use it the way it was built back then in combination with TX816 modules where the A, B and A + B switching with different channel sounds really shines. As soon as i will get the serial midi port working i can report some infos regarding channel aftertouch, as the KX88 is able to do so. For now i am still trying to get one of my cheap chinese plastic housing built Yamaha KX (KX49 and KX25) from the 2000er years recognized and working. For some strange reason there is no midi data in debug mode showing up, and no sound because of that...."

_Originally posted by @StephanBeer in https://github.com/probonopd/MiniDexed/discussions/162#discussioncomment-2630192_

eLuSiVeMiTe, 2022-04-25:
Is it class compliant.  (Will it run on windows with no drivers) 
Idle curiosity.

probonopd, 2022-04-26:
Hello @StephanBeer, please remove `usbspeed=full` from the `cmdline.txt` on the SD card and try again.

StephanBeer, 2022-04-26:
Hey @probonopd and @eLuSiVeMiTe, first off thanks for your work on that wonderfull and exciting project !!

I removed the usbspeed line so that cmdline.txt is blank, but that did not change anything, there is still no midi data incomeing and received from the KX keyboard. Messages from the Triple Play are still received and show up on the screen, PCs and Volume via Fishman Controller working fine.

When putting the same sdcard from the Pi Zero into the Pi 1B, it seems that the Debug Monitor is frozen and no messages are showing up, like when unplug or replug the controller. Is the problem for the Raspeberry Pis resolved ?? Thats with the newest build...

I would love to have a "confirmed working with following USB devices" section in the WIKI someday, which would be helpful to exclude possible errors when building the project. Right now its often hit and miss for me, so its hard to do a systematic debugging to find the reason when things won´t work.

eLuSiVeMiTe, 2022-04-26:
> Hey @probonopd and @eLuSiVeMiTe, first off thanks for your work on that wonderfull and exciting project !!
I'm just along for the ride. Wouldn't know where to start with programming. (But I could build you a house) 

I had a good dig last night and I can't find any mention of that controller being class compliant (runs on standard drivers) 
Im assuming midi input devices need to be class compliant to work?

Try using a usb midi interface and midi din from the Yamaha.

StephanBeer, 2022-04-26:
> @StephanBeer I'll ask again, is that device MIDI class compliant?

I have been using the Yamaha KX USB keyboards for about 14 years now, on Windows XP, Windows 7, 8 and now 10. As i switched to Linux (for music, recording and composing Windows is stil needed) the last 2 years, the Yamaha KX are gladly showing up in every application or with pipewire, but i guess thats because they are present in the Linux Kernel ? To cut it short i never have tried to use them without a driver. I have to remove the drivers and plug on of the keyboards in and see if they are working without, or even show up in the device manager.... are there other ways to find out USB class compliance ? Plug them into ones android phone ?

StephanBeer, 2022-04-26:
> 

Plugged the keyboard via USB type B to micro USB to USB C adapter into my android phone, and it is enabled only as a physical keyboard....a keyboard on whih you type, not a midi keyboard. Normally there should be a dialog popup where you can change to midi device ?

StephanBeer, 2022-04-26:
> 

> > Hey @probonopd and @eLuSiVeMiTe, first off thanks for your work on that wonderfull and exciting project !!
> 
> I'm just along for the ride. Wouldn't know where to start with programming. (But I could build you a house)
> 
> I had a good dig last night and I can't find any mention of that controller being class compliant (runs on standard drivers) Im assuming midi input devices need to be class compliant to work on the pi
> 
> Try using a usb midi interface and midi din from the Yamaha.

It really seems that its not Class compliant, so i will have to switch the Output Mode of the keyboard from USB to the traditional MIDI ports and finish building the serial MIDI ports on my MiniDexed build.