probonopd/MiniDexed #182: Internal mixer mixes linear and not logarithmic

dcoredump, 2022-04-27:
... as the subject says... I think panningbis also linear. Should it be also logarithmic?

Will fix this soon for internal mixer class (if noone else wants to do this, because I am currently writing the SYSEX engine).

probonopd, 2022-04-27:
Great chatch! According to https://sound.stackexchange.com/a/41550,

> The MIDI specification itself does not define the exact response to the volume controller.
> 
> Nowadays, there is almost no device that does no implement the expression controller. Both the GM Level 1 Developer Guidelines and the DLS specification use L(dB) = 40 log (volume × expression / 127²). Furthermore, the guidelines mention that
> 
>> there was general agreement about how these two controllers interacted as well: In 9 cases, their values were combined (multiplied) to get the actual level. The recommended volume response curves for CC#7 (volume) and CC#11 (expression) used herein were provided to the General MIDI Working Group of the IASIG/MMA by Yamaha Corporation. Roland uses the same response curve, and other Japanese manufacturers who are members of the AMEI have agreed to do the same.
> 
> So this is how almost all devices actually implement it.

probonopd, 2022-05-11:
Merged. Thank you @dcoredump

closed 2022-05-11 by probonopd