probonopd/MiniDexed #185: Add more detail on use of voice sysex files in main "Usage" instructions

diyelectromusic, 2022-04-28:
Although links are provided to voice ROMs in the "Features" section of the readme, I think it would be useful to add a little more detail to the "Usage" section to cover the fact that they need to be renamed into the bank_name format and placed in a sysex/voice directory.

This wasn't clear (at least to me anyway) when trying to get started with MiniDexed and if this is wrong, then there is no indication on the LCD, just a blank screen.  The error message shown on HDMI implies the files are in the wrong format, not that they have the wrong name - I ended up going to the code to work out what was wrong. It also wasn't clear to me at that time, that the numbering was related to the bank number, so to change "roms" I had to select a new bank via the menus (as an aside, being able to scroll off the end of one bank onto the next using the encoder would make selecting voices a lot easier too).

I know the provided script sorts this out if used, but for anyone creating an SD card on a Windows PC that probably won't work for them and I wonder that just mentioning the voice file naming convention would help people get started more easily.

Very many thanks for MiniDexed, it is already providing a lot of fun.  I've played with MT32-Pi and Circle myself, so I'm very interested in getting more "up to speed" with what you've done here too :)

probonopd, 2022-04-28:
Hello @diyelectromusic, would you like to suggest some text for https://github.com/probonopd/MiniDexed/wiki/Files?

diyelectromusic, 2022-04-28:
> Hello @diyelectromusic, would you like to suggest some text for https://github.com/probonopd/MiniDexed/wiki/Files?

Ah yes, I can do.  Do I just edit the wiki page directly?  Also, isn't it "sysex/voice" not "sysex/voices"?  I think that caught me out at one point...

Then presumably a note can be added to the main readme about the format of voice files?

probonopd, 2022-04-28:
Yes `sysex/voice/000000_rom3a.syx`

https://github.com/probonopd/MiniDexed/blob/main/getsysex.sh

closed 2022-04-28 by diyelectromusic