probonopd/MiniDexed #186: Weird switch issue

romix123, 2022-04-28:
Hi, I have built a miniDexed with a ky-040, lcd1602 and pcm5102 board on a raspi 2b
Everything works, except the switch. Encoder moves through the menu, but button press does nothing, at least most of the time. Occasionally a press is registered but I can’t figure out when or why.
The switch is ok (press to zero), tried with separate switches, wire to ground, moving input to other gpio, nothing works.
Do I miss something, or is this a known issue (on a 2b)?

craigyjp, 2022-04-28:
Are you double clicking as required?

romix123, 2022-04-28:
Wow, this is embarrassing. Double clicking does the trick.

don’t be like Ronald, rtfm.

thanks

craigyjp, 2022-04-28:
It's ok, I thought the same for a few minutes after building mine. A quick read of the manual fixed it for me.

There is a request to reverse it so one click is enter and double click is back, which is what I would prefer.