probonopd/MiniDexed #198: LCD screen displaying random characters or just black squares

dlawle, 2022-05-03:
Hello, 

I’m running into an issue where upon the first connection, my lcd display showed this:
[image]

After a power cycle, it only displayed a single row (top) of black squares (all matrix grid filled). 

I'm running on a pi 4, with a 128gb sd card, followed the tutorial in the linked video. I’ve tried a pot between VO and that didn’t display anything either. Could this be my pi, or could my sd card be too large?

dlawle, 2022-05-03:
I believe this may belong in discussion. Please move if so!

craigyjp, 2022-05-03:
Try powering the display from  3.3v.