probonopd/MiniDexed #203: README: Document advantages over running on Linux

dronus, 2022-05-04:
While I am thrilled by this project, I wonder what the benefits over running Dexed on Linux on the Raspberry would be. 

Maybe the documentation should say something about that. As there are of course a lot of benefits running  some Linux OS. 

What comes to my mind is of course much lower power-on latency, and most likely much better audio latency. But how much? What else?

dcoredump, 2022-05-07:
Another advantage (for me the most important advantage!): The whole thing still works in 10 or 20 years, even if your future OS no longer offers driver support. In addition, you are not subject to constant updates and the risk that something will not work afterwards. This is incredibly important if you want to use the synth on stage. Nothing is worse than when your musical instrument stops working 1 hour before the performance due to problems with the OS!

probonopd, 2022-05-07:
* Faster startup time
* Lower latency
* Way easier and faster installation and updates
* Fewer external dependencies
* No need to configure JACK
* More direct control over the specific Raspberry Pi hardware
* Lower poerformance overhead
* Easier reuse of existing Synth_Dexed and MicroDexed code
* Overall much simpler
* Less like a "computer", more like an "instrument"
* Not developing against a "moving target" (Linux distribution)

If you are looking for Linux based solutions, there are already existing ones, including Zynthian. A graet project at what it does, not as easy to configure, not as slim, and not as fast.