probonopd/MiniDexed #214: Serial Midi In double triggering notes

miotislucifugis, 2022-05-11:
Just discovering that with the last few builds (from today) midi In via Serial is causing a double trigger of notes... both on Note On (when key is pressed)  and Note off (key is released)... eventually this causes the pi/ minidexed to stop responding altogether.
Perhaps this may be why Ive been having difficulty sending sysex parameter changes.

I just installed an older version of MiniDexed (before sysex stuff was added) and it behaves normally

probonopd, 2022-05-11:
Does this one work?
[MiniDexed_2022-05-05-76620d1](https://github.com/probonopd/MiniDexed/suites/6392503597/artifacts/232482733)

It is the build from 76620d1.

miotislucifugis, 2022-05-11:
Sure thing!

[MiniDexed_2022-05-05-76620d1](https://github.com/probonopd/MiniDexed/suites/6392503597/artifacts/232482733) - yes working

[MiniDexed_2022-05-09-3894cb3](https://github.com/probonopd/MiniDexed/suites/6433675941/artifacts/235332344)
no 

[MiniDexed_2022-05-11-b6df892](https://github.com/probonopd/MiniDexed/suites/6458091795/artifacts/236945096)
no
I didnt bother trying the last one

probonopd, 2022-05-12:
Looks like it might have to do with https://github.com/probonopd/MiniDexed/pull/195 then?

cc @dcoredump

dcoredump, 2022-05-12:
> Looks like it might have to do with https://github.com/probonopd/MiniDexed/pull/195 then?
> 
> cc @dcoredump 

Umpf... Have to take a deeper look at this. I hope my scope is arriving today, so I can fix my MiniDexed. After Superbooth I will have some more time for this and fix it!

diyelectromusic, 2022-05-12:
I'm not entirely sure, but I notice that m_SerialMIDI.Process() is called twice from CMiniDexed.Process()?

And there might also be something odd in the RunningStatus handling in CSerialMIDIDevice.Process as a consequence of the SysEx additions.  m_nSerialState/case 4 has lost it's else statement that reset MIDI handling in the case of receiving a CMD message (it has moved to cases 1,2 which are already processing data bytes...)... but this might be a "red herring" as I'm getting a little lost following the two goto statement's logic...

dcoredump, 2022-05-12:
> I'm not entirely sure, but I notice that m_SerialMIDI.Process() is called twice from CMiniDexed.Process()?
>
> And there might also be something odd in the RunningStatus handling in [...]

Yes, you are totaly right! Thanks for reporting.will fix this in an hour.

I think this happend due to my excessive debugging when implementing SYSEX until I found that the problems was my MIDI input circuit...

probonopd, 2022-05-12:
Thanks @dcoredump

closed 2022-05-12 by probonopd