probonopd/MiniDexed #215: Microtuning support

lukexi, 2022-05-12:
Would be an incredible addition! 

Looks like Dexed has it, so I guess it's a matter of porting it into Synth_Dexed and then into MiniDexed?

dcoredump, 2022-05-12:
Hi @lukexi ,

I thinks MicroTuning is not very difficult to implement, but currently I have much others tasks in my list. 

<Advertisement>
I can only call on everyone to try to deal with the programming (of MiniDexed). If you know a little C/C++ and have the patience to learn the code, you will find a very interesting world. In addition, there is here (and also on other forums) help. For me in particular it would be extremely helpful, if I would not have to work alone on the Synth_Dexed library. Then also feature requests could be worked in faster 😁.
</Advertisement>

diyelectromusic, 2022-05-19:
> [Advertisement] I can only call on everyone to try to deal with the programming (of MiniDexed).

I'm interested in getting more involved, but as is probably apparent, but Github skills are still in the "getting started" mode, and my C++ is rudimentary at best.  But my background is embedded C so, if you can give me a few pointers, I'm happy to have a go!  I've also not done any Teensy yet... but I have some... :)