probonopd/MiniDexed #230: Compiling problems with I2C display

dcoredump, 2022-05-16:
This is what the current main branch produces for me:
```
userinterface.cpp: In member function 'bool CUserInterface::Initialize()':
userinterface.cpp:71:45: error: no matching function for call to 'CHD44780Device::CHD44780Device(CI2CMaster*&, unsigned int&, const unsigned int&, const unsigned int&)'
   71 |        CConfig::LCDColumns, CConfig::LCDRows);
      |                                             ^
In file included from userinterface.h:26,
                 from userinterface.cpp:20:
../circle-stdlib/libs/circle/addon/display/hd44780device.h:69:2: note: candidate: 'CHD44780Device::CHD44780Device(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, unsigned int, boolean)'
   69 |  CHD44780Device (unsigned nColumns, unsigned nRows,
      |  ^~~~~~~~~~~~~~
../circle-stdlib/libs/circle/addon/display/hd44780device.h:69:2: note:   candidate expects 10 arguments, 4 provided
../circle-stdlib/libs/circle/addon/display/hd44780device.h:55:7: note: candidate: 'constexpr CHD44780Device::CHD44780Device(const CHD44780Device&)'
   55 | class CHD44780Device : public CDevice /// LCD dot-matrix display driver (using HD44780 controller)
      |       ^~~~~~~~~~~~~~
../circle-stdlib/libs/circle/addon/display/hd44780device.h:55:7: note:   candidate expects 1 argument, 4 provided
make: *** [../circle-stdlib/libs/circle/Rules.mk:144: userinterface.o] Fehler 1
```

I havn't found any matching constructor in `./circle-stdlib/libs/circle/addon/display/hd44780device.h`

luisfcorreia, 2022-05-16:
I'm having the same issue

I've read somewhere this weekend that minidexed used Circle's dev branch where this change (for I2C) was made

diyelectromusic, 2022-05-16:
Yes that's right, you need to go into circle-stdlib/libs/circle and switch branches to develop.  At least that is what I did I think!  I'm still a bit of a github newbie myself...

probonopd, 2022-05-16:
Yes. Need to run:

```
           cd circle-stdlib/libs/circle
           git checkout c9a4815 # develop
           cd -
```