probonopd/MiniDexed #244: Some performances sound garbled on RPi 2

probonopd, 2022-05-21:
Some performances sound garbled on the Raspberry Pi 2, e.g.,
* Default `performance.ini` with 4 Tone Generators
* All 4 TGs set to voice `STRG ENS` or `TUB ERUPT`

This does not happen on RPi Zero 2 WH, 3, or 4.
It also does not happen with most other voices on the RPi 2.

CPU overload?
Would distributing the workload to the cores differently help?
Or is the Raspberry Pi 2 reaching its limits here?

matrix12x, 2022-05-23:
I use this on a RaspberryPI 2. Which bank are these 2 patches in? (So I can try and replicate the bug.)
I have not had this issue with any patches(on 4 tone gens set to same patch same channel) thus far.

probonopd, 2022-05-24:
`rom3a.syx` according to https://yamahablackboxes.com/collection/yamaha-dx7-synthesizer/patches/.