probonopd/MiniDexed #246: Throttle CPU to lower temperature

QuadratClown, 2022-05-22:
I've recently built a header for a Pi 4 with the minidexed components, however to save space and make the result flatter, the pi4 is mounted upside down. Even though a heatsink is mounted on the cpu and the device is not mounted into a case, a few sustained tests showed that the PCB is running quite hot - even too hot to touch after a while.

Since the kernel calls `m_CPUThrottle.Update ();`, I suspected that `socmaxtemp=45` in circle's `cmdline.txt` would limit the CPU speed and throttle it down, however that is not the case - it still ran equally hot.

Are there any other ways, e.g. by throttling the CPU or undervolting, to lower the temperature without having to resolve into reworking the hardware? I'm quite happy with the design hardwarewise, would like to avoid having to resolve to active cooling or rework the pi4 mount.

sou7611, 2022-05-22:
This is where one of those aluminum cases that expose the GPIO pins comes in very handy - turning the entire case into the heat sink can make a huge difference.

probonopd, 2022-05-22:
I would imagine that the options in `config.txt` can be used for underclocking/undervolting:

https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking-options

Also see https://github.com/rsta2/circle/issues/140#issuecomment-705651000.

cc @rsta2

rsta2, 2022-05-22:
`socmaxtemp=NN` has an effect, only when `fast=true` is specified at the same time. But you shouldn't do this, because it would make the situation even worse, and 45 degrees Celsius SoC temperature should not be maintainable.

I think mounting the RPI 4B upside down is not a very good idea for cooling. I'm not an expert for cooling, but from the software side there is nothing, which can be done to reduce the temperature further.

QuadratClown, 2022-05-22:
@rsta2 I only tested the extreme setting of 45 degrees to see if anything changes at all.

I agree, the upside down mount is certainly not the ideal way of mounting the pi. I just love how flat and convenient the result is. Since the HD44780 displays are so thick, it's hard to mount it on top of the pi without raising the height of the module.

It would certainly be doable with a tightly intregrated design on PCB and utilizing and/or using Pi Zero 2, however the former requires more development time and the latter is almost impossible to purchase.

@sou7611 That is a great idea. I might be able to reach someone with access to a CNC machine, that would certainly be an awesome case. However, that's something for future me :D

FYI, I managed to get the temperature in check by using a Pi 3 instead. It runs hot still, but way less so than the Pi 4.

For reference:

[image]

@probonopd I'm not too familiar with the bare metal implementation of circle, do options in `config.txt` like cpu frequency still apply? The `CPUThrottle` class of circle seems to be able to control everything cpu related, hence I'm not sure which settings apply and which are ignored/overwritten.