probonopd/MiniDexed #250: Configurability of input methods

QuadratClown, 2022-05-22:
The default way of navigating the menu is very convienent in the sense that it only requires a single push-encoder, however it is somewhat unintuitive and lacks in comparison to e.g. multi-button options in terms of navigation speed.

The main speed-bump is the menu back-and-forth navigation.

Ideas:
-  Currently, a single tap goes back and a double tap goes forth. I found it very inconvient at first and would have expected it to be the other way round. How about adding an configuration option to flip these?
- Since back and forth are distinct actions, how about configuring two additional GPIO's to perform them? If they are NC, nothing will change, but i would allow for two dedicated buttons in additiona, which would greatly improve the usability of the device standalone.

probonopd, 2022-05-22:
> Currently, a single tap goes back and a double tap goes forth. I found it very inconvient at first and would have expected it to be the other way round.

Why would you have expected it the other way around, are there instruments that use a single click to open and a double click to leave a menu?

We are considering/discussing https://github.com/probonopd/MiniDexed/pull/155 but I think it'd be very hard to explain and might confuse users.

> Since back and forth are distinct actions, how about configuring two additional GPIO's to perform them?

The encoder button could be made the "OK/Enter" button and a secondary button could be made the "Back/Cancel" button. But it would require more hardware, and complicate the build, so it should be optional.

What do you think about this:

* Make arbitrary buttons on MIDI controllers usable as the "OK/Enter" button and as the "Back/Cancel" button

QuadratClown, 2022-05-22:
> Why would you have expected it the other way around, are there instruments that use a single click to open and a double click to leave a menu?

I'm used to gear that relies havily on encoder-driven menu-diving, however with no double-click action, but a `back` option in the menu instead. So a single click is always `Enter` or `forwards`. One example is the Squarp Pyramid Sequencer. Others use the same approach as well, the MiniDexed is the only piece of gear I know that changes the basic `click->enter` relationship. Additionally, I know of no gear at all that uses double-click for something as essential as `enter`, instead it was only - if mapped at all - some kind of shift or mode-switch behaviour.

> Make arbitrary buttons on MIDI controllers usable as the "OK/Enter" button and as the "Back/Cancel" button

MIDI Controllers as in external controllers, e.g. connected via USB? That would already be a step in the right direction ...

> The encoder button could be made the "OK/Enter" button and a secondary button could be made the "Back/Cancel" button. But it would require more hardware, and complicate the build, so it should be optional.

... but I would prefer this (as in I would prefer this over MIDI only, but both would be perfect). I agree that it absolutely should be optional. The minimal viable build with a single encoder makes a lot of sense.

luisfcorreia, 2022-05-22:
I would vote for a "back" option instead of double-click

matrix12x, 2022-05-22:
I would also vote for a "back" option instead of double-click

probonopd, 2022-05-24:
@Banana71 that should already be working exactly like you describe - does it not work for you?
https://github.com/probonopd/MiniDexed/wiki/Menu#shortcuts

Banana71, 2022-05-24:
@probonopd Those who can read have a clear advantage. Shortcuts :-D It's magic. I have to read the fucking manual. But i think it was a nice idee. Thx

diyelectromusic, 2022-08-23:
There have been a number of new button configuration options added to MiniDexed now.  Has the original issue now been solved sufficiently that this can be closed?

probonopd, 2022-08-25:
I think so. Please comment here if you think this needs to be reopened. Thanks!

closed 2022-08-25 by probonopd