probonopd/MiniDexed #262: Cosmetic: 'mididevice: Send SYSEX voice dump to <garbled>'

probonopd, 2022-05-29:
Cosmetic error in the HDMI log:

`mididevice: Send SYSEX voice dump to <garbled>`

I think this appeared as part of
https://github.com/probonopd/MiniDexed/pull/195

Can you have a look please @dcoredump? 

https://github.com/probonopd/MiniDexed/blob/c00ed093984bc4eb60ffa1806170eb95e5400df2/src/mididevice.cpp#L475

dcoredump, 2022-05-30:
I think it should be a `LOGDGB()` instead of a `LOGNOTE()`.  It is shown because at startup (or better: when loading a voice) a voice dump is send on this channel, so external editor software listening on that channel has the actual stae of the voice. This was really convinient for MicroDexed (with max. 2 TGs on different channels) but perhaps not for MiniDexed?

Should this behaviour be configurable or perhaps removed?

probonopd, 2022-05-30:
I see, seems reasonable to me. [Maybe Dexed will support it one day?](https://github.com/asb2m10/dexed/discussions/352)

The only "complaint" is a cosmetic one: the `<garbled>` part of the message.

dcoredump, 2022-05-31:
> I see, seems reasonable to me. [Maybe Dexed will support it one day?](https://github.com/asb2m10/dexed/discussions/352)
> 
> The only "complaint" is a cosmetic one: the `<garbled>` part of the message.

Ahh, ok. Will try to fix this and send a PR.

dcoredump, 2022-05-31:
> I see, seems reasonable to me. [Maybe Dexed will support it one day?](https://github.com/asb2m10/dexed/discussions/352)
> 
> The only "complaint" is a cosmetic one: the `<garbled>` part of the message.

Ahh, ok. Will try to fix this and send a PR.

> I see, seems reasonable to me. [Maybe Dexed will support it one day?](https://github.com/asb2m10/dexed/discussions/352)
> 
> The only "complaint" is a cosmetic one: the `<garbled>` part of the message.

This is now fixed in #265

probonopd, 2022-06-05:
Thanks @dcoredump, I confirm that it is working beautifully now :+1:

closed 2022-06-05 by probonopd