probonopd/MiniDexed #282: Midi Out will fail when a 0x0A character is in the output.

BenZonneveld, 2022-06-14:
In the file circle-stdlib/libs/circle/include/circle/serial.h the defaults translate a NL to NL+CR thereby changing the size of voice dumps and transmitting a panning ctrl which also is 0x0A (NL)

// serial options
#define SERIAL_OPTION_ONLCR     (1 << 0)        ///< Translate NL to NL+CR on output (default)

diyelectromusic, 2022-06-16:
Interesting!  Good spot :)

So it looks like there ought to be a call to m_Serial.SetOptions with a value of 0 (there doesn't seem to be a definition for "clearing options") in serialmididevice.cpp for CSerialMIDIDevice::Initialize do you think?

I can see where the default setting of options is to the above in circle/serial.cpp... but there doesn't seem to be a way to user-set this without using SetOptions that I can see?

BenZonneveld, 2022-06-16:
I fixed it by changing the default in that file as I saw no other options.
At least my sysex out is working correctly now

probonopd, 2022-06-19:
Could you please send a pull request/diff? Thanks @BenZonneveld

diyelectromusic, 2022-08-23:
@BenZonneveld does the proposed fix fix this for you now?

Many thanks,
Kevin

probonopd, 2022-08-25:
Merged #311 for now so that the fix is in the latest continous builds for easy testing; please let me know if the merge causes any trouble for you. Thanks!

diyelectromusic, 2023-01-22:
I propose that this issue is now closed as this has now be implemented and as far as I know has solved the problem unless told otherwise.

closed 2023-01-22 by probonopd