probonopd/MiniDexed #299: Support for adafruit i2c / spi backpack

Nbobito, 2022-07-01:
I recently finished my minidexed, but the LCD screen did not light up when I went to power it on. I first checked all my connections, and they were in the correct place. I decided to hook the LCD up to an Arduino and run a [demo](https://learn.adafruit.com/i2c-spi-lcd-backpack/arduino-i2c-use), which verified that it was working. I then checked that my `minidexed.ini` file was set up right with the correct i2c address and all that, and it was. Also, before the backpack was put on, I had it directly connected, which also worked. So that rules out any hardware issues.

I think the issue is the Adafruit backpack requires that the LCD be turned on first, using `LCD.setBacklight(HIGH);`, otherwise, it remains dark. This is from the Arduino demo, but it should also be doable on the raspberry pi. 

Other people might have bought this backpack for the minidexed as well, as it is the first google search result for ["i2c backpack,"](https://www.google.com/search?q=i2c+backpack) so it would be nice if support for this backpack could be added.

probonopd, 2022-07-02:
Do you know which i2c command needs to be sent to the backpack as the equivalent for `LCD.setBacklight(HIGH);`?

Nbobito, 2022-07-03:
I beleive so. The function calls _digitalWrite, which in this case sends an i2c command.
```cpp
// Allows to set the backlight, if the LCD backpack is used
void Adafruit_LiquidCrystal::setBacklight(uint8_t val) {
  if(val) // no dimming so any non zero value turns on backlight
        val = HIGH;
  // check if i2c or SPI
  if ((_i2cAddr != 255) || (_SPIclock != 255)) {
    _digitalWrite(7, val); // backlight is on pin 7
  }
}
```
This code is from the Adafruit [LiquidCrystal](https://github.com/adafruit/Adafruit_LiquidCrystal/blob/master/Adafruit_LiquidCrystal.cpp) library, which is made for Arduino, but I'm assuming it could be done on a Raspberry Pi.

probonopd, 2022-07-05:
"backlight is on pin 7" - that means that the Backpack is using a separate pin for the backlight?

diyelectromusic, 2022-07-05:
I've also seen some other displays that use I2C for backlight control - there are some with an RGB backlight, so you have to tell it which colours to turn on... (I think DFRobot did one of these).

But there are quite a number of variants, so when I was looking at I2C LCDs, it really wasn't clear how to support these variants without having special code for each new one I found :(

Of course, one option might be a hardware mod to disable the programmability of the backlight...

diyelectromusic, 2022-08-23:
Support for different displays, especially those with additional configuration needs, is an issue for the upstream Circle environment really - so can this issue be closed here?

probonopd, 2022-08-25:
Would be good if an owner of the "adafruit i2c / spi backpack" could confirm. @Nbobito did you get it to work?

Nbobito, 2022-08-30:
I tried it with the most recent release and could not get the LCD to work. I could be setting it up wrong, or it could be an issue with the Circle environment. Either way, I'm okay if this issue issue closed.

probonopd, 2022-09-03:
No one on the dev team has a adafruit i2c / spi backpack, so it is probably not working with MiniDexed and is unsupported. If anyone wants to send a pull request to add support for it, that'd be very welcome. Thanks!

closed 2022-09-03 by probonopd