probonopd/MiniDexed #301: LCD display freezes or doesn´t refresh with pi pico circuitpython usbMIDI matrix keyboard

Sebizarro, 2022-07-03:
I've manage to build pi pico midi matrix keyboard from an old Yamaha pss-102 toy piano, it works great with Dexed on the PC, but with the minideXed (rpiZero2) it freezes or doesn´t refresh the LCD, mainly when changing voice or bank. It still changes the voice but no feedback from the LCD.
I also tried with the keystep from arturia and it works great. With the midimonitor all the data looks ok from the keyboards.
Don't know if it is a problem with minidexed or the pico matrix keyboard.

https://github.com/diyelectromusic/sdemp/blob/master/src/SDEMP/CircuitPython/PiPicoToyMIDIMatrixDecodeUSBUART.py

diyelectromusic, 2022-07-05:
You want to be careful with any code they publish, I'm really not sure they really know what they're doing ;)

Seriously, there is an issue with CircuitPython USB MIDI endpoints that is a little odd.  Have you seen the comments on cmdline.txt options?  See: https://github.com/probonopd/MiniDexed/wiki/Files#minidexedini I didn't think it would make it freeze, but it means MiniDexed doesn't recognise it...

The alternative solution is a special boot.py file that disables the extra endpoints (again as described in the above link).

Kevin

Sebizarro, 2022-07-05:
Thanks Kevin.

I've notice that the micropython normal midi was easier to follow but i don't know much about coding and didn't found a USBMIDI version for that. Still having lots of fun with both the diy keys, minidexed and with zynthian, the pc worked more or less ok until i started to mess with the code.

Tried to add a encoder and a small oled display to change the midi channel, have some feedback info on notes and so on but soon gave up. just got to the bouncing ball scene and some lag on the USBmidi.

Will check the boot.py file solution, but probably will change for micropython and try to add a normal midi to both the DIY midi keyboard and the minidexed.

here is one of the enclosures i did for the minidexed (raspberrypi zero 2, small DAC, i2c LCD, encoder).

Thanks again

sebastião

[MiniDexed_Enclosure_RPIzero2_V1.zip](https://github.com/probonopd/MiniDexed/files/9050114/MiniDexed_Enclosure_RPIzero2_V1.zip)

diyelectromusic, 2022-08-23:
Hello @Sebizarro - did you manage to get anywhere with this? Is this still an open issue?

closed 2022-09-17 by probonopd