probonopd/MiniDexed #308: No sound if OP6 is disabled (Enable=OFF)

arsamus, 2022-07-14:
I get a total mute behavior when I set Enable=OFF on OP6. Does anyone else have this issue? It doesen't happen on Desktop Dexed.

Banana71, 2022-07-15:
Hello, I have tried some sounds and can confirm the error with algorithms 16 -18. Any other algorithm will work if OP6 is disable.

Incidentally, the OPs remain disabled after a program change.

arsamus, 2022-07-15:
Hi @Banana71 ok, I will have a look.

probonopd, 2025-04-20:
@Banana71 can you reproduce this behavior on Dexed on the PC?

probonopd, 2025-04-21:
So how do you Enable/disable the OPs when it misbehaves? Via the menu and/or via Sysex?

Banana71, 2025-04-21:
I build a Performance OP-Test, copy to SD-Card Performance Laboratory 003:011 [000011_OP-Test.zip](https://github.com/user-attachments/files/19837435/000011_OP-Test.zip)

In the performance, TG1 features a voice with six different frequencies, from OP1, which is low, to OP6, which is very high. If individual OPs are disabled or enabled, you can clearly hear that the assignment is incorrect. It's even clearer if all OPs are disabled first (Enable=Off) and then enabled individually (Enable=On).

[image]

probonopd, 2025-04-21:
And you are disabling/enabling operators using (standalone, not VST) Dexed you say?

That does't even seem to have ANY impact on the MiniDexed continuous build... I can change algorithms but seemingly cannot switch on/off operators, and do many other things (such as detune).

Banana71, 2025-04-21:
> And you are disabling/enabling operators using (standalone, not VST) Dexed you say?

No, the miniDexed can only enable or disable the OPs directly in its own UI. Dexed cannot control the OPs in the miniDexed.

Banana71, 2025-04-22:
Perfect, the switches are now assigned to the correct operators.

But after changing a voice or performance, the operators aren't reactivated. Dexed reactivates all operators when loading a different voice.

soyersoyer, 2025-04-22:
I also included the fix for that in #870.

Banana71, 2025-04-22:
Very good, now the operators are reactivated after a voice or performance change. Even after a voice MIDI dump via Dexed, deactivated operators are reactivated.

I don't know if @arsamus is still reading this, but this post can be closed.

Thank you very much.

closed 2025-04-22 by probonopd