probonopd/MiniDexed #309: MAX98357A Adafruit module tested and working but a little quiet

danriches, 2022-07-15:
Hi All,

I didn't have a PCM5102 or PCM5122 board so I hooked up a MAX98357A that I had lying around and it does work although rather quietly with a 3W speaker attached and the gain turned up to 15dB (add 100K resistor from Gain pin to GND). It sounds quite clean but due to the volume being too low I'll be ordering a PCM5102 board and hooking this into a class D amp for more oomph!!

Great job by the way to everyone involved in the project, I'm loving playing with this! It took me about an hour to wire up a 20x4 LCD, bare rotary encoder with pullup resistors, classic MIDI DIN socket and the MAX98357A amp. This was probably the easiest but most satisfying build ever and it worked first time, kudos to you all!!!

Dan

probonopd, 2022-07-15:
Thanks @danriches for testing the MAX98357A, I added this information to https://github.com/probonopd/MiniDexed/wiki/Hardware#i2c-dac.

diyelectromusic, 2024-01-31:
If this is not an issue, can this report be closed now?

Kevin

closed 2024-02-01 by probonopd