probonopd/MiniDexed #314: "Send to all" button?

hizzlekizzle, 2022-07-18:
While I love having all of the duplicate instances for the rich, fat tone, it makes it a hassle to change things like voices on the fly, and it would be nice to be able to send a setting to all of the instances at once. I figured it could work as one of the assignable functions for the pins/buttons as mentioned in [the button section of the wiki](https://github.com/probonopd/MiniDexed/wiki/Files#buttons).

However, I understand that the "shortcut" function (rotary press-and-turn) is intended to fulfill much the same functionality, so it won't hurt my feelings if you guys don't consider this worth the additional effort. :)

Unrelated, is the STL file for [this case](https://user-images.githubusercontent.com/2480569/169115034-0be57842-79f9-4c41-a8c4-9db9e43f62c1.png) available somewhere? I didn't see it linked anywhere on the wiki.

probonopd, 2022-07-18:
This could be implemented as a menu item, and it would copy the settings for the current Tone Generator (TG) to all TGs.

But maybe there is an even better, simpler way:

Looks like, e.g., the Korg opsix has a __Unison__ mode, where you set up one voice, and then just enable Unison with three parameters: How many voices to use (2...8), Spread (how much Stereo Pan to use), Detune (how large the overall range for is for Detune across all voices). See the Unison, Detune, Spread sections in the Korg opsix manual. WIth more powerful Raspberry Pis, maybe we can have more than just 8 voices in total? 8 TGs wich can have Unison would theoretically give us up to 64...

hizzlekizzle, 2022-07-18:
Yes, I think that would do the trick, as well!