probonopd/MiniDexed #336: Overlapping pin configurations can generate weird error messages

diyelectromusic, 2022-08-16:
We probably need some configuration sanity checking on the range of possibilities for GPIO button configurations now.  In particular, if the rotary encoder is configured with duplicate GPIO pins by mistake, there is an assert failure that isn't very easy to track down.  It might be quite handy to eliminate (or at least warn about) other potential clashes with "fixed pins" - e.g. with I2C, the LCD, I2S, MIDI (depending on the other settings of course).

I suspect we ought to do some checking on the buttons in general.

Aside: I was also wondering if it was possible to have two hardware buttons configured for the same job?  I was initially thinking of a board that allows either the rotary encoder switch for select, but also a separate hardware button for select too, but having dropped the RE switch configuration, this is no longer possible.

I know this is now somewhat contradictory to my above comment about checking for duplicates! :)

Kevin

BobanSpasic, 2022-08-16:
I know that this is a promoting of my software, but here you have some sanity check:
[image]

As soon as some pin is assigned, it is removed from the selection list for other pins.

miotislucifugis, 2024-09-09:
I realize that this is a closed issue, but I would second the request for having 2 buttons that can do the same job (or configuration).  Also, if possible, a single job that is assigned to 2 buttons (and configuration) ... basically allowing a button to act as a "shift" or "alt".   for example,  if a button is assigned to "performance up" (on a "click"),  it could/ would do something else (TG up, etc)  when it was pressed again (click) while holding the  "shift" button  (long press).