probonopd/MiniDexed #338: Might be nice if scrolling through voices automatically moves to next bank

diyelectromusic, 2022-08-23:
When scrolling through voices on one voice bank, then it might be useful to be able to configure the system so that the next voice bank is automatically selected when then end of one bank is reached, with wrap-around for the last bank back to the first one.  If you see what I mean...

arsamus, 2022-08-24:
Yes. It would be a good functionality since it is a bit disgusting go back to select bank over and over in order to find the desired voice... 

I could do it, maybe as an optional function setted on Minidexed.ini I'll push a PR soon.

probonopd, 2022-08-25:
Indeed. It would be welcome.

If I am not mistaken, then @abscisys either had implemented or wanted to implement this in a fork. I think it _might_ even be hidden somewhere in https://github.com/probonopd/MiniDexed/pull/293/files... not sure though. 

@abscisys had [let us know](https://github.com/probonopd/MiniDexed/pull/293#issuecomment-1167732539) that
> Now that I am a Project Manager or even program director of IT projects, I do no longer work on Conf Management system.

diyelectromusic, 2022-12-10:
Here is my suggested fix for this: https://github.com/diyelectromusic/MiniDexed/tree/Issue338Fix

Grateful if anyone free could have a quick look and see what they think.

Kevin

probonopd, 2022-12-10:
Never forget to send pull requests :+1: 

https://github.com/probonopd/MiniDexed/pull/392

probonopd, 2022-12-10:
The advantage of doing a PR is that we get a testable build. Here it is:
[MiniDexed_2022-12-10-11472af](https://github.com/probonopd/MiniDexed/suites/9789395492/artifacts/470798087)

Banana71, 2022-12-10:
Works perfectly, even the jump from the last bank to the first. Well done

closed 2022-12-11 by diyelectromusic