probonopd/MiniDexed #356: Some trouble with Classic MIDI In and old Roland Piano

Banana71, 2022-09-26:
Hello,

i have some troubles with the classic midi input

after releasing a single note, the Roland RD-300 or the master keyboard Roland MKB-300 sends the command "CC: All Note off". MiniDexed then promptly cuts off all tones. The sustain pedal also has no effect. Other devices such as Waldorf Blofeld or the Roland JX-08, JD08 can handle it and can be played well with the RD-300.
Is there a way to filter the "CC All Note Off" command in MiniDexed.

Logfile vom Roland RD-300:
[image]

If it ever works I'll put the MiniDexed in the RD-300 and MKB-300.

probonopd, 2022-09-27:
According to https://community.cantabilesoftware.com/t/roland-d-50-midi-cc-123-all-notes-off/2157/6, certain Roland devices send CC 123 ("All Notes Off") when the last key is released. This is probably a bug or a misinterpretation of the MIDI standard by Roland at the time.

Apparently Roland owners just filter out CC 123.

We would need to make this configurable via `minidexed.ini`, since it is not desirable for non-Roland owners.

Banana71, 2022-09-28:
Thank you very much Holger,
It's time to learn how to develop.

diyelectromusic, 2022-09-28:
Interestingly this is a channel mode message, so only tone generators listening on a specific channel should respond to it.  The MIDI spec says that any instruments in Omni mode should ignore All Notes Off (see "channel mode messages: all notes off".  Also synths don't have to respond to it, it is optional...

So we ought to have something in the tone generators that checks for omni mode and ignore it if set...

probonopd, 2022-10-03:
Please test [MiniDexed_2022-10-03-dc1a0e0](https://github.com/probonopd/MiniDexed/suites/8591707243/artifacts/384912002)

Set `IgnoreAllNotesOff=1` in `minidexed.ini`.

Banana71, 2022-10-03:
It works very well, only sometimes there is hanging a tone. Maybe it's another problem with the old Roland hardware. Tomorrow I will take a little more time and investigate with MIDI-OX, among other things. 
Thank you very much

probonopd, 2022-10-03:
Hanging tones could also be an issue on the Raspberry Pi software side. Which model of RPi are you using?

Banana71, 2022-10-03:
I use a Raspberry Pi 3 Model B+

probonopd, 2022-10-03:
Does it happen when you put a lot of load on the machine?

Banana71, 2022-10-05:
I've played 4 TG's simultaneously with high polyphony on other Masterkeyboards connected via USB or classic MIDI In. So i exclude the Raspberry Pi in the problem.

Banana71, 2022-10-09:
After I deleted the line "usbspeed=full" in the "cmdline.txt" file, it ran even more stably. 

> Please test [MiniDexed_2022-10-03-dc1a0e0](https://github.com/probonopd/MiniDexed/suites/8591707243/artifacts/384912002)
> 
> Set `IgnoreAllNotesOff=1` in `minidexed.ini`.

I would be very happy if you would apply your changes to "Main".

closed 2022-11-25 by Banana71

probonopd, 2022-11-25:
Thanks, merged.

PR #366 (2022-11-25): Implement IgnoreAllNotesOff=1
Implement `IgnoreAllNotesOff=1` in `minidexed.ini`, closes #356

According to https://community.cantabilesoftware.com/t/roland-d-50-midi-cc-123-all-notes-off/2157/6, certain Roland devices send CC 123 ("All Notes Off") when the last key is released. This is probably a bug or a misinterpretation of the MIDI standard by Roland at the time.

Apparently Roland owners just filter out CC 123.