probonopd/MiniDexed #369: ChannelsSwapped seems to be non-functional

ramphands, 2022-10-15:
As mentioned in the Discussion section by numerous people, there seems to be an issue with the "ChannelsSwapped" variable - it doesn't switch channels regardless of being set to 1 or 0.

Did it previously work and has been broken along the way of development?  I'm no coder, but I tried to find all the instances of it being called and it all seems fine.

Could some other users test to see if the functionality works on their hardware?
I'm using the purple GY-PCM5102 DAC board which seems like the most common variant being used for this project.

probonopd, 2022-10-15:
I had only tested this functionality with HDMI.

ramphands, 2022-10-16:
> I had only tested this functionality with HDMI.

Ahh, I see.  Is there any way to get it to work with the DAC boards?  I guess I could do some trace cutting on the boards, but a software solution would be far more elegant.  

Thank you for your hard work by the way - I love this project!

diyelectromusic, 2022-12-01:
I've had a quick look and, well, its complicated... but I can't see where it is going wrong atm.

What I know:
* There is a HW specific output format, which depends on the xxxSoundBaseDevice in use (I2C, HDMI, PWM).
* There is a "write" output format which is set to signed 16-bit 2 channel, except on a RPi V1 when it is 16-bit 1 channel.
* In most cases the output code has to call a convert function to get from one to the other that depends on pointer adjustments, sizeof sample size, etc...
* The Circle SoundBaseDevice has the concept of "channels swapped" too, but it is different to the use of the ChannelsSwapped configuration parameter.
* As far as I can see, only the PWM sound base device uses Circle's ChannelsSwapped parameter, and then only for certain versions of the RPi.  This does mean that setting ChannelsSwapped in the config for PWM output will swap it twice!
* I've added debug code for the handling of the ChannelsSwapped config variable, and it appears accurate everywhere it is used.
* As far as I can see the code managing the left/right indexes in ProcessSound is set correctly.

One thing I did wonder... there are a number of places where sizeof(s16) for SoundFormatS16 is used and s16 is typedef'd to short, but I seem to recall that short is one of those types that has a definition of "at least 16 bits" so wondered what size it is on a Pi and if that could cause odd things to happen where it is assumed that this is two bytes...?  But I'm grasping at straws a little now, as I can't spot why this isn't working!  Unless there is something weird in the TGs themselves when handling stereo...

I suppose what I'm saying is that the audio path is complicated and dependent on the version of Pi, the DAC/output (of course), and the configuration... but that the code for e.g. handling the configuration has many different paths depending on the Pi/DAC, and there are several layers - in the Dexed library, MiniDexed source, and Circle...

I'll keep looking...

diyelectromusic, 2022-12-01:
Ok, I think this is the problem...

The Dexed engine I believe is mono, so it returns a single channel of audio.  I think the "stereo"-ness gets applied in the "effect_mixer", using the routine (https://github.com/probonopd/MiniDexed/blob/6d14da86040dea4819e205952cb8c970854b320c/src/effect_mixer.hpp#L120):
    void doAddMix(uint8_t channel, float32_t* in)

There is another version of this function that would take two inputs (L,R) but that isn't used in minidexed - as I say, Dexed has a mono output.  So this function takes the same input buffer and applies the panning controls.

The the mix is obtained later on using the getMix function (https://github.com/probonopd/MiniDexed/blob/6d14da86040dea4819e205952cb8c970854b320c/src/effect_mixer.hpp#L155):
    void getMix(float32_t* bufferL, float32_t* bufferR)

So whatever goes in the first parameter gets the left channel and whatever goes in the second parameter gets the right channel.

So this is where things go wrong (https://github.com/probonopd/MiniDexed/blob/main/src/minidexed.cpp#L951):

		// swap stereo channels if needed
		uint8_t indexL=0, indexR=1;
		if (m_bChannelsSwapped)
		{
			indexL=1;
			indexR=0;
		}

Yet when getMix is called:
			tg_mixer->getMix(SampleBuffer[indexL], SampleBuffer[indexR]);

And the sample buffers are used later to store samples:
					tmp_float[i*2]=SampleBuffer[indexL][i];
					tmp_float[(i*2)+1]=SampleBuffer[indexR][i];

... it doesn't matter if indexL is 0 or 1 - indexL will still always point to the "left" channel...?

What do people think?

The fix seems to be to either to change the getMix line to the following so the original "Left" is always 0:
			tg_mixer->getMix(SampleBuffer[0], SampleBuffer[1]);

That way changing indexL between 0 and 1 will really swap channels...

OR to check the ChannelsSwapped later swap L/R just prior to actually storing the buffers in tmp_float[]...?  Say around here: https://github.com/probonopd/MiniDexed/blob/main/src/minidexed.cpp#L1005

What do people think?  This seems to fix it from my simple tests so far...

ramphands, 2022-12-03:
I'll have a go with it in the next couple of days and will report back.

ramphands, 2022-12-03:
@probonopd Just to confirm, with the updated install, should I set channelsswapped to 1 or just install as is?  Not sure whether the install is using the first or second version of @diyelectromusic 's fixes.

diyelectromusic, 2022-12-03:
> @probonopd Just to confirm, with the updated install, should I set channelsswapped to 1 or just install as is? Not sure whether the install is using the first or second version of @diyelectromusic 's fixes.

Ideally have a go with ChannelsSwapped = 0 first then set it to 1 and see if it makes a difference?  This is a fix for recognising the ChannelsSwapped configuration setting. Whether you need it for your specific set up i'm not sure :)

Banana71, 2022-12-04:
Moin, 
now it works exactly as it should, good job
(ChannelsSwapped tested with the purple GY-PCM5102 DAC board)

mongio, 2022-12-05:
Is the fix already included in latest build? [MiniDexed_2022-12-03-d03717d.zip] 'Cause I've just installed and checked but nothing changes in my configuration (channels still swapped).

mongio, 2022-12-05:
> Is the fix already included in latest build? [MiniDexed_2022-12-03-d03717d.zip] 'Cause I've just installed and checked but nothing changes in my configuration (channels still swapped).

I answer myself: no, but the test version up here works a charm :-) (PCM5102 DAC, swapped channels problem solved - need to swap to 1 in my system to get the right positioning - thanks!!)

diyelectromusic, 2023-01-22:
I believe this issue can now be closed as completed.

closed 2023-01-22 by probonopd