probonopd/MiniDexed #379: MiniDexed cannot handle more than 40 performances

Banana71, 2022-11-25:
There are 39 Perf.ini files in my performance folder. When I try to save a new performance, the Raspberry Pi crashes.
I've tried a few things with no success. (ID sorted, more than 40 performances in the folder)

[performance org.zip](https://github.com/probonopd/MiniDexed/files/10089372/performance.org.zip)
[performance 68.zip](https://github.com/probonopd/MiniDexed/files/10089371/performance.68.zip)

Make a back up from your data before copying my performance to the SD card.

probonopd, 2022-11-25:
[This](https://github.com/probonopd/MiniDexed/pull/228/files#r1032076741) looks suspicious.

probonopd, 2022-11-25:
Hi @Banana71. Can you please try, on a **test** miniSD card (that contains no important files), [MiniDexed_2022-11-25-961f76c](https://github.com/probonopd/MiniDexed/suites/9500389176/artifacts/450706221)?

Banana71, 2022-11-25:
@probonopd It works. 1024 performances should be enough for now :-)

probonopd, 2022-11-25:
Just ping me again when you are reaching the limit. Then we can increase it again :+1:

commit 2bd6f411 (2022-11-25): Increase m_nPerformanceFileName, closes #379
closed 2022-11-25 by probonopd