probonopd/MiniDexed #384: Initial brass sound stays throughout all patches

Slider2732-2, 2022-12-02:
A perplexing situation.
Build is a Pi 4, with a rotary encoder, latest MiniDexed but no screen or DAC yet. My Beatstep Pro didn't seem to fire notes, nor Roland JU-06A. Instead, am using a USB qwerty keyboard, due to no dedicated MIDI controller keyboard.
Also, the sysex/voice folder is present and sounds are in there from Dexed cart.
At boot, I can play a brass type of sound.
On pressing the rotary switch twice (slowly) I can then dial in other sounds by moving the encoder anti-clockwise.
However, the brass sound stays and can also be heard along with any new sound.

The brass sound can be lowered, by setting all 8 pan values to zero in performance.ini, but it's still present. 
I tried swapping out the sounds from sysex/voice to other ones and still the brass sound plays.
Next thing, completely deleting the sysex/voice folder.. aha, the initial bank is within the main code!

Though, it can't be a bug if noone else has the problem?
It must be because of having to use a USB qwerty keyboard?

Many thanks for any help :)

Banana71, 2022-12-02:
Hi Slider2732-2,

the default performance.ini is a layer of four Brasses TG1 to TG4. You can set in Performance.ini the MIDIChannel2, MIDIChannel3 and MIDIChannel4 to "0" than you hear only a single Sound in TG1.

[performance TG1 only.zip](https://github.com/probonopd/MiniDexed/files/10138833/performance.TG1.only.zip)

with pan you can set the left-right balance in the stereo-output.

probonopd, 2022-12-03:
> I remember having the same issue when first trying MiniDexed. The default performance.ini might be interesting to show the potential, but it isn't very "beginner friendly". I do wonder that the default ought to be a lot simpler?

Yes, the idea of the default performance is to show the potential of the device. For "real work", a different performance is preferable.

diyelectromusic, 2022-12-03:
> Yes, the idea of the default performance is to show the potential of the device. For "real work", a different performance is preferable.

In that case, as someone has already got as far as installing MiniDexed at this point then I think the default performance.ini should be the simplest to allow them to explore it further.  As pretty much the first thing people will do is try to change the voice I think that ought to work with no surprises "out of the box" :)

But maybe that is just me!

probonopd, 2022-12-03:
Well, my thinking was that it needs to sound impressive out of the box. Once people start editing voices, they probably understand how to change/edit a file :)

diyelectromusic, 2022-12-03:
Well, I'd like to think I had a pretty good handle on things like this, but when I installed MiniDexed for the first time it was pretty confusing for me to start with, trying to work out what I'd done wrong as patch changes didn't seem to be having much effect... I wasted a fair bit of time before realising that several channels were all sounding at once and it wasn't actually a problem with my build or (main) configuration.

The fact that it has been raised as an issue means I wasn't the only person perplexed as to what was going on.  I guess it depends on the mental model you're bringing to MiniDexed. I'd started with seeing it as a TX816, so 8 independent "synths" i my mind, so I hadn't spotted that several were actually playing at the same time, hence the confusion.

I still believe that if someone is going to get to the point of putting together enough of a system to actually try MiniDexed for themselves, then they are probably already sold on the idea of having a DX7. I don't think they'll need convincing any further :)

As far as "editing a file" goes - actually no, that means taking your SD card to a PC and fiddling around with it, but the UI encourages changing "within" the system. In many cases I expect the first action for many is to set tone generators to independent MIDI channels and configure pan settings as something fairly generic so there is a "blank" baseline to start from, but happy to be told I'm wrong.

As we have the option of saving performance settings now, I think it makes more sense to start with a "blank slate" and let people fiddle via the UI than start with a complex setup that they have to "decode" (and undo) before doing their own thing?

But I'm only throwing my comment in as a "me too" having this issue, so I'd be interested in hearing others thoughts :)

Maybe start with a "simple" and a "showing off" performance.ini in the GitHub and make people choose one on installation so at least it is obvious this is something to consider?

I tend to work on the "principle of least surprise" for things.  For a multi TG MiniDexed, least surprise for me says the first install out of the box should let you change the voices and hear them change :)

I understand how it all works now, so its not an issue for me, but I'm just adding my voice here it as I think there will be some new users getting caught out by this one.

Kevin

Banana71, 2022-12-03:
**We have everything we need. :-)**

In my opinion, the default performance should also be a simple single patch. In addition, the sysex/voices/... and the performance/.. structure should already be created, incl. some Performances.

[MiniDexed Sounds and Performances.zip](https://github.com/probonopd/MiniDexed/files/10146946/MiniDexed.Sounds.and.Performances.zip)

The sysex files are from Yamaha Black Boxes
https://yamahablackboxes.com/collection/yamaha-dx7ii-fd-synthesizer/
https://yamahablackboxes.com/collection/yamaha-dx7-synthesizer/patches/

Is it possible to start the miniDexed directly in the "Performance - Load" menu?

probonopd, 2022-12-17:
> Is it possible to start the miniDexed directly in the "Performance - Load" menu?

This is not yet implemented but should be doable.

diyelectromusic, 2023-01-22:
Did we resolve this?  I think the proposal is for a "simple" performance.ini to be the default to minimise surprises for new users?

I don't know how this relates to the new performance save/load options though.

Kevin

diyelectromusic, 2023-04-01:
Bumping this.  Can we change performance.ini to a single TG responding to MIDI OMNI for "getting started" reasons, especially now there are many performance files to choose from thanks to @Banana71 ?

Kevin

probonopd, 2023-04-01:
@diyelectromusic my line of thinking is that the out of the box experience ("what I hear out of the box without any configuration") should show the capabilities of the device. More advanced users are more likely to know how to load a blank INIT as the default performance than new users trying to get a sound out of the device for the very first time. @Banana71 it is clear that someone like you wants to change the default, and that's easy to do, isn't it?

diyelectromusic, 2023-04-01:
I know that is what you think, but I maintain that the most common thing for a new user to do when turning it on is to change the voice and see what they sound like, and having the thing respond as if it is ignoring them isn't very friendly, as evidenced by this issue, by this also being raised in discussions elsewhere, and it mirrors my own first experience with MiniDexed too - it took me quite a bit of "head scratching" to realise what was going on - remember at this stage I could have got so much wrong bulding it - missing ROMs, getting the audio output wrong, - it took a while to spot that all TGs were enabled in OMNI mode still playing the default sound, and I'd like to think I was pretty clued up on what it was doing :)

If they've already installed it to the point where they have it running, then they don't need convincing of its capabilities and at the end of the day there is so much out there about the DX7 already.  But giving them a problem to solve right out of the box for something so basic as changing the voice feels like it can put people off really quickly.

Having changing the voice only working for "more advanced users ... who know how to load a blank INIT" kind of makes the point for me I'd say.  And out of the box the thing will already make a sound as it even has a default full bank of voices to load if there are no ROMs.  But when something as simple as changing the voice doesn't work out of the box, I think that is a mistake.

I suspect we won't see "eye to eye" on this one :)

But either way, regardless of what you decide, this issue should probably be closed!

Cheers!
Kevin

Banana71, 2023-04-01:
As with everything, there are different approaches and that's a good thing.
My thoughts on this:
The manufacturers of the synthesizers that I know don't start their instruments with an INIT sound. Newcomers who start the miniDexed for the first time should be offered the best possible listening experience. After the system start, the miniDexed should start in the Performance Load menu, so that everyone can easily switch between the performances.

A small QuickStart guide in the form of a video or a one-pager showing the first steps would be helpful. I've also been thinking about a FAQ page on Wiki.

> @Banana71 it is clear that someone like you wants to change the default, and that's easy to do, isn't it?

I don't understand exactly what you mean now? Who wanted what?

probonopd, 2023-04-01:
Sorry if I was not clear, @Banana71 I just used you as an example of a power user who wants to program his own voices and probably had no difficulty to change the default performance to something else.

diyelectromusic, 2023-04-01:
> The manufacturers of the synthesizers that I know don't start their instruments with an INIT sound.

They offer a bank of preset voices you can select between and hear working.  MiniDexed can do that too, but the default performance setting means that selecting a new voice is still sounding Brass 1 on three TGs making it sound like voice selection isn't working.  I know of no manufacturer that gives you an "out of the box" experience that looks like changing patches doesn't work :)  If you push a button on a DX7 to select a different voice, it changes the voice, it doesn't keep sounding Brass 1.

We're not talking about a blank INIT voice or bank, we're talking about a simple performance configuration that allows the user to change voices and be able to hear those changes take effect.

Anyway, I'm repeating myself now :)

Kevin

probonopd, 2023-04-01:
Yes, I can see both sides. I'd say, let's see how many complaints we get about the current default.

Would it be preferable to have the performance selection (rather than the voice selection) more prominent in the menu?

diyelectromusic, 2023-04-01:
Well three comments have surfaced on this issue that I'm aware of, and I only mentioned it as a "me too".  It wasn't a complaint, it wasn't worth mentioning otherwise, but it did cost me quite a bit of wasted time when I got started trying to work out why it wasn't doing what seemed the most obvious way to work.  I always work on the "principle of least surprise" on things like this.  What person won't be surprised when they change a voice and it doesn't seem to work?

I think as most people will come at this as a "DX7 on a Pi" rather than a "TX816 on a Pi", and more people would be familiar with a single multi-timbral synth than layering synths (how many people had a TX816 back in the day compared to the DX7?) then I would say the performance side of things should remain in the "more advanced" category until people are familiar with how to drive a single TG.  It is much simpler in my view to explain how the "single synth TG" works and then tell people they can have up to 8 than start with a multi-layered, multi-timbral, polyphonic, detuned setup and try to explain how all that works enough to allow them to wind it back.

A "real instrument bought in a store" will not ignore you when you change the voice ;)

Kevin

Slider2732-2, 2023-04-01:
Thanks very much for continuing the discussion, it is very much respected.

Perhaps the resolution really is just a simple note of explanation added to the install guide.
Banana71's zip could be linked to for those with minimal experience or equipment?
It should, in my opinion, be stressed that it's only an entry level introduction when configured in such a way.  
The properly configured code stays as is, others with no skills can know that they are being catered for.

My own experience is thought to be in line with most 'casual' tinkerers and musicians, the use of a Pi being strongly linked by people to 'hacking' something to emulate something else. I've no doubt the USB keyboard being used to trigger notes would be viewed similarly. 'We' cludge when not owning the proper equipment for the job at hand. 
Personally, I don't like the idea that someone may try out MiniDexed with a USB keyboard, hit this newbie problem and go off to make a weather station with their Raspberry Pi!

probonopd, 2023-04-01:
> Perhaps the resolution really is just a simple note of explanation added to the install guide.
Banana71's zip could be linked to for those with minimal experience or equipment?

Yes, in fact, we are shipping Banana71's performances with MiniDexed now...

Slider2732-2, 2023-04-01:
Ah, wonderful!

Issue resolved  - thank you probonopd :)

diyelectromusic, 2023-04-04:
> Well three comments have surfaced on this issue that I'm aware of,

Make that four (probably): https://github.com/probonopd/MiniDexed/discussions/472

Kevin

diyelectromusic, 2023-04-06:
(five https://github.com/probonopd/MiniDexed/discussions/472#discussioncomment-5547432)

diyelectromusic, 2023-06-05:
(six https://github.com/probonopd/MiniDexed/pull/500#issuecomment-1576176557)

diyelectromusic, 2024-07-30:
(seven https://github.com/probonopd/MiniDexed/issues/664)

diyelectromusic, 2024-12-11:
(eight https://github.com/probonopd/MiniDexed/issues/772#issuecomment-2530537222)

diyelectromusic, 2024-12-11:
> Can we change performance.ini to a single TG responding to MIDI OMNI for "getting started" reasons, especially now there are many performance files to choose from thanks to @Banana71 ?

This is what I'd like to see in the default performance.ini. I think it will cause the least issues for a new user. It is an easy config to start with and a pretty easily understandable one to go back to after messing around with other performances.

All in my view of course.

Kevin

Banana71, 2024-12-12:
Hello @diyelectromusic Kevin,

I agree with you, except for the embedded voice data. 
I like to be surprised but when I call up a performance I also want to get what I expect.
Without VoiceData in the default performance the sound depends on the voice bank on the SD card. In my miniDexed the first voice on bank 1 is a kick drum. When I then play a chord after starting the miniDexed I am really awake. :open_mouth:

I have now done the following:  [performance.ini.txt](https://github.com/user-attachments/files/18106719/performance.ini.txt)
TG 1 = OMNI; all other TG = disabled. TG1 with the Brass 1 VoiceData, All other TG have no voice data.

With a fresh system with or without voice banks there are no surprises. Without existing voice banks the miniDexed behaves like a DX7 and loads the voices from ROM1A. After saving a performance the voice data is there anyway.

Peter

Banana71, 2024-12-14:
The language barriers make it difficult to understand complex topics without misunderstandings.

> I only meant with the original performance, not with all.
- Yes, I always only spoke about the default performance.ini.

> I'm just after "what is least surprising for a new user, who doesn't know their way around the system yet" setup for someone following the instructions we have online.
- I want to achieve the same thing. Where did we describe which sound comes when you first switch it on? I can't find anything in the WIKI. Wasn't that only explained in the discussion area?

> I don't have a strong opinion I guess on embedded Brass 1 vs bank 1 voice 1. I'm still inclined towards the latter, as weren't there international DX7s with different ROM sets? Bank 1, voice 1 on startup feels "right" to me rather than hard-coding a sound they might not have even installed...
- The confusion does not come from the default sound but rather from the fact that it is so difficult to set a different sound due to the complexity of 8 TGs.
- If there is no sysex/voice data on the SD card, ROM1A in the miniDexed is active as bank 1. The first sound on ROM1A is the "Brass 1" patch.

Perhaps a short guide explaining the first steps (voice selection, load performance, save performance, save default performance) would help.

@probonopb can replace the performance.ini as a test if he wants, there is always a way back or we can think of something better.

Peter

diyelectromusic, 2024-12-14:
I guess I just assumed the original DX7 would power up with bank 1, voice 1 selected.
I'm pretty sure my DX11 does that, but I don't have a real DX7 to test :)

Kevin

probonopd, 2024-12-14:
It's a tradeoff:
* 1 voice by default, or
* Show the full power and beauty of this instrument

diyelectromusic, 2025-02-21:
The more I think about this, the more I think we do need a "no performance setup, it's just 8 sets of banks and voices" configuration and the basic performance.ini seems like the obvious place to do that.

It is the least surprise for a user when they first switch it on, and complex sounds that really show off the whole instrument are just a couple of performance settings away, which IIRC is basically the same number of menu selections as getting to voice anyway (especially as you can turn "TG1->performance" with a single "left" turn).

So how about we provide a performance.ini that does the following:
* Map each TG to the same numbered MIDI channel.
* Turn all effects/vol/pan/etc to default values.
* Does not include any embedded voice and assumes "bank 1, voice 1" for all TGs (which will still exist even if no sysex directory).

This is kind of like the "INIT VOICE" of performances then and allows it to be used as a multi-timbral 8-voice, polyphonic tone generator over MIDI, but is easily configured for either voice/bank or performance/bank MIDI PC messages via minidexed.ini or of course, can quickly be changed via the menu.

Kevin

Banana71, 2025-02-23:
I also think that it should be made as easy as possible for beginners.
What we are missing, however, is documentation that simply explains the first steps for the different applications of the miniDexed. (Performance player, multitimbral, sound design, ...)

I have created a performance that meets all the expectations of the miniDexed and can actually be supplied on its own, without any additional performance. 
Put yourself in the position when you started the miniDexed for the first time and play this performance.

[miniDexed - Initial performance.zip](https://github.com/user-attachments/files/18929307/miniDexed.-.Initial.performance.zip) Unzip and copy into the root directory of your SD card. (You may want to back up the existing performance.ini first)

I would be delighted to hear your feedback!

diyelectromusic, 2025-02-23:
That looks great, but I think it isn't quite getting the point I was trying to make. I think we need a "MiniDexed isn't in performance mode" performance.ini as the root/default.

So on powerup, it is in "8 tone generator with voices" mode, like 8 default DX7s (which after all is what the system is advertised as).

But with a selection of any other performance it is a multi-timbral, 8-tone generators-as-voices-themselves, instrument.

But there will always be that one performance setting that someone can go back to which is "8 individual tone generators" for when that is required.

To do this, the performance would have to be as I described it above.

The best way to explore an instrument like this, I strongly believe, is to have both:
* Some great default patches, easily selectable, that really show it off - which we have in the performances.
* A known, default starting state for further exploration on one's own.

We have the former, but we don't have the latter as the default performance has effects configured, several TGs responding all to OMNI, and some not responding at all - so it isn't in a known, understandable state for a new user.

I also think that having the latter makes the former appear even more impressive as you get an immediate comparison between "blank slate MiniDexed" and "crafted sonic experience MiniDexed" and more importantly, as a new user, you can understand why - i.e. you've just selected an action and have observed its effect :)

Kevin

Banana71, 2025-02-24:
@diyelectromusic Kevin, I knew what you were talking about.

[INIT-Multitimbral performance.zip](https://github.com/user-attachments/files/18936451/INIT-Multitimbral.performance.zip)

```
#
# performance.ini (INIT-Multitimbral)
#

# TG#
#BankNumber#=0		# 0 .. 127
#VoiceNumber#=1		# 1 .. 32
#MIDIChannel#=1		# 1 .. 16, 0: off, >16: omni mode
#Volume#=100		# 0 .. 127
#Pan#=64		# 0 .. 127
#Detune#=0		# -99 .. 99
#Cutoff#=99		# 0 .. 99
#Resonance#=0		# 0 .. 99
#NoteLimitLow#=0	# 0 .. 127, C-2 .. G8
#NoteLimitHigh#=127	# 0 .. 127, C-2 .. G8
#NoteShift#=0		# -24 .. 24
#ReverbSend#=0		# 0 .. 99
#PitchBendRange#=2 # 0 .. 12
#PitchBendStep#=0 # 0 .. 12
#PortamentoMode#=0 # 0 .. 1
#PortamentoGlissando#=0 # 0 .. 1
#PortamentoTime#=0 # 0 .. 99
#VoiceData#= # space separated hex numbers of 156 voice parameters. Example: 5F 1D 14 32 63 [....] 20 55 
#MonoMode#=0 # 0-off .. 1-On
#ModulationWheelRange#=99 # 0..99
#ModulationWheelTarget#=1 # 0..7
#FootControlRange#=99 # 0..99
#FootControlTarget#=0 # 0..7
#BreathControlRange#=99 # 0..99
#BreathControlTarget#=0 # 0..7
#AftertouchRange#=99 # 0..99
#AftertouchTarget#=0 # 0..7

# TG1
BankNumber1=0
VoiceNumber1=1
MIDIChannel1=1
Volume1=100
Pan1=64
Detune1=0
Cutoff1=99
Resonance1=0
NoteLimitLow1=0
NoteLimitHigh1=127
NoteShift1=0
ReverbSend1=99
PitchBendRange1=2
PitchBendStep1=0
PortamentoMode1=0
PortamentoGlissando1=0
PortamentoTime1=0
VoiceData1=
MonoMode1=0 
ModulationWheelRange1=99
ModulationWheelTarget1=1 
FootControlRange1=99 
FootControlTarget1=0 
BreathControlRange1=99 
BreathControlTarget1=0 
AftertouchRange1=99 
AftertouchTarget1=0 

# TG2
MIDIChannel2=2
# TG3
MIDIChannel3=3
# TG4
MIDIChannel4=4
# TG5
MIDIChannel5=5
# TG6
MIDIChannel6=6
# TG7
MIDIChannel7=7
#TG8
MIDIChannel8=8

# Effects
#CompressorEnable=1	# 0: off, 1: on
#ReverbEnable=1		# 0: off, 1: on
#ReverbSize=70		# 0 .. 99
#ReverbHighDamp=50	# 0 .. 99
#ReverbLowDamp=50	# 0 .. 99
#ReverbLowPass=30	# 0 .. 99
#ReverbDiffusion=65	# 0 .. 99
#ReverbLevel=80		# 0 .. 99

# Effects
CompressorEnable=1
ReverbEnable=1
ReverbSize=68
ReverbHighDamp=20
ReverbLowDamp=20
ReverbLowPass=80
ReverbDiffusion=65
ReverbLevel=22

```

probonopd, 2025-02-24:
Can we make it so that by default (first time experience), MiniDexed makes a "powerful" sound (using 8 TGs and effects and everything), but the `INIT-Multitimbral` performance is very easy to access? Maybe we should write a section about it in the documentation?

diyelectromusic, 2025-02-24:
It already has a bank of very powerful sounds that are very easy to access.

What it doesn't have is an /intuitive/ starting point from which a new user (who won't immediately know where all the menus are to wind back things like detune, pan, reverb, etc) can experiment with things like effects and multitimbrality, hence my suggestion.

It would be a bit like having a PSS/PSR keyboard from the 80s that always starts up playing the "demo" when actually you just want to get on and use the thing.

But at this point I've said all I can to convince, and despite the evidence here that this still causes confusion for new users, we're just going around in circles :)

So I'm probably going to leave it there until someone else reports they are having the same problem.

Kevin

lezarith, 2025-02-28:
This beginner setup is grueling, ngl. I've spent hours fiddling with setting, reading, changing lines, and so on. After finally getting my screen to work and the encoder button presses to do work, I'm at the stage where I'm trying to get the banks I dl'd from blackboxes to work after doing the whole "numbers_name" bit. 

Could there not be an easier to follow tutorial out there, or an updated pre-configured group of files to make this is easier and less time consuming?

probonopd, 2025-02-28:
Hello @lezarith.

If you want to play just one voice, then you need to disable all but one Tone Generators (TGs) by setting them to other MIDI channels than the one you are using, and load the voice you like into that TG. https://github.com/probonopd/MiniDexed/blob/main/getsysex.sh iwill put the sysex files into the correct directory for you.

Hope this helps!

We are always open for improvements in the documentation; which points would you like to have seen where in the documentation?

diyelectromusic, 2025-02-28:
I've added a note to this: https://github.com/probonopd/MiniDexed/wiki/Hardware#pinout
to detail which things in the default minidexed.ini file need to change to support that starter setup.
Note: the encoder is supported and enabled in the default minidexed.ini file as per those diagrams.

Kevin

Banana71, 2025-03-01:
> I don't mind sinking some time building the device and learning what it can do, but spending hours pouring over the documentation and debugging does kill the enthusiasm.

For a few weeks now, I have found it very difficult to separate fun and seriousness from each other 🧱 

[image]