probonopd/MiniDexed #387: Compilation warnings of out of order initialisation for minidexed.cpp

diyelectromusic, 2022-12-09:
As title.  Initialisation of member items for CMiniDexed::CMiniDexed is generating compiler warnings as below due to mismatch between ordering in minidexed.cpp and minidexed.h:

```
In file included from minidexed.cpp:20:
minidexed.h: In constructor 'CMiniDexed::CMiniDexed(CConfig*, CInterruptSystem*, CGPIOManager*, CI2CMaster*, FATFS*)':
minidexed.h:300:7: warning: 'CMiniDexed::m_bSavePerformance' will be initialized after [-Wreorder]
  300 |  bool m_bSavePerformance;
      |       ^~~~~~~~~~~~~~~~~~
minidexed.h:262:7: warning:   'bool CMiniDexed::m_bSavePerformanceNewFile' [-Wreorder]
  262 |  bool m_bSavePerformanceNewFile;
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~
minidexed.cpp:33:1: warning:   when initialized here [-Wreorder]
   33 | CMiniDexed::CMiniDexed (CConfig *pConfig, CInterruptSystem *pInterrupt,
      | ^~~~~~~~~~
```

probonopd, 2022-12-09:
I don't know how to fix this, maybe you have an idea?

diyelectromusic, 2022-12-10:
This is what I've done - reordering items in minidexed.h seemed the easiest.  I don't think there are side effects of doing this, but can't exactly be sure.  Can someone review?

https://github.com/diyelectromusic/MiniDexed/tree/Issue387Fix

closed 2022-12-11 by diyelectromusic

probonopd, 2022-12-12:
Thanks @diyelectromusic