probonopd/MiniDexed #396: Ad a "All TG" menu

probonopd, 2022-12-11:
Add a "All TG" menu in addition to TG 1...8.

Rationale:
Sometimes we want to change an aspect of all TGs at once (e.g., the voice).

Banana71, 2022-12-11:
like this idea.

What do you think of this menu?
I/O for changing certain TG's

Multi TG Edit 1-8
X X O O X O O X

The settings would change TG 1, 2, 5 and 8

probonopd, 2022-12-12:
Would be good for power users but probably also be more complicated to use and to code.

@diyelectromusic wdyt?

diyelectromusic, 2022-12-12:
> @diyelectromusic wdyt?

I must admit, I wasn't entirely sure of this idea at all.  Is this related to the discussion elsewhere about a "unison mode"?  And what do you do if you select "all TG" but the parameter you are changing is different for each one - do they all converge on the same parameter?  If numerical, do they all increment from their own starting point by 1?  This only really makes sense for certain options I'd say, like changing voice.  But if you are wanting all TGs to play the same voice at the same time, then they'd have the same MIDI channel, and so would all respond to a PC message over MIDI?

I guess what I'm saying is that this is probably possible with the right MIDI configuration already, so do we need to try to mimic it in the UI too?  Or if we do, then the idea of configuring a single "unison mode" instrument would probably make more sense to me - so all TGs become "one" as part of some top-level "global setting".  But then I guess even then there would have to be some parameters that wouldn't be the same - otherwise what is the point (e.g. detune)?

I don't know.  I certainly think that "all TG" is probably complicated enough to understand the implications of, so any kind of subset of selected TG editing is getting really, really complicated so personally think I'd recommend we don't go there...

I'm inclined, in general, to treat MiniDexed a bit like the original TX816 - you'd really need to drive it over MIDI for the full power.  MiniDexed already has a much more sophisticated UI compared to three buttons and a 7-segment display, but I think if it gets too complicated then it will become unusable...

Summary: I can see a "make the whole thing act like one 8-way TG" unison mode being useful, but we'd need to think about certain parameters that might need to remain unique, but I'm still not sure about "all TG" but might be convinced... but I really think "some selectable subset of TGs" is too far.

That's my current thinking, but very happy to hear opinions :)

abscisys, 2022-12-13:
Hello,

If I may, mimicking what Roland proposed as editing capabilities in its D / JV / XV... series is good if you have quick access buttons to enable / disable TGs and select / deselect TGs hence a more complete physical controls set.

Personally, I would be on the same page as @diyelectromusic and would rather expect the All TG menu being activated if the Unison mode is enabled as it would greatly ease the use of the Unison.

Enabling Unison mode would:
+ Clone TG1 to all other TGs: TG2 to TG8
+ All TG menu entry appears before TG1 or alternatively replaces all menu entries TG1 - TG8
+ Detune param of All TG does not set the tuning of the TGs but rather represents the detuning spread across all TGs
+ Pan param of All TG does not set the Pan of all TGs but rather represents the panning spread a across all TGs

Disabling Unison mode would:
+ Reload the latest loaded performance

abscisys, 2022-12-19:
Hello,

I have some code ready for a proposition for both:
[x] #315: Unison mode 
[x]  #396: All TG menu

Shall I propose to send a pull request so someone can have a look into it and provide a feedback? Still I do not really what to do when disengaging the Unison mode.

In terms of XP, it adds a menu as a sibling to Effects named "Unison" that contains:
* Enable: true/false
* All TG: for a global edit of all parameter of all TGs. 2 exceptions: Pan and Detune that both represent the spread of respectively the Detune and Pan params. Note that when enabling Unison, this is TG1 that serves as a template for the other TGs 2 to 8 so that appart from Detune and Pan all the other params of TG1 are copied over to other TGs 2 to 8