probonopd/MiniDexed #397: SSD1306 display misaligned when SoundDevice=i2s

pegli, 2022-12-13:
My MiniDexed build has been working great with PWM output.  I added a PCM5102 DAC to my HAT, which is providing much better sound.  When I enable it by setting SoundDevice=i2s in minidexed.ini, the SSD1306 display often shows a warped version of the menu or no menu at all:

[image]

Switching back to SoundDevice=pwm fixes the behavior.

Banana71, 2022-12-13:
Hi @pegli,

can you copy your minidexed.ini here?
I would check the following lines in the minidexed.ini:

# Default is 16x2 display (e.g. HD44780)
LCDColumns=20
LCDRows=2

Maybe an address conflict?!
The DAC you use usually works perfectly.

pegli, 2022-12-13:
Here are my 1306 settings:

```
# SSD1306 LCD
# For a 128x32 display, set LCDColumns=20; LCDRows=2
# For a 128x64 display, set LCDColumns=20; LCDRows=4
SSD1306LCDI2CAddress=0x3c
SSD1306LCDWidth=128
SSD1306LCDHeight=32

# Default is 16x2 display (e.g. HD44780)
LCDColumns=16
LCDRows=2
```

I also have `DACI2CAddress=0`, and both the DAC and the display are functioning correctly except for the misalignment.

I was also thinking that this might be an address conflict or a problem with DAC address detection.  I'll boot the Pi into a desktop and see if I can find the address of the DAC, and maybe set it manually to see if that helps.

diyelectromusic, 2022-12-13:
Hmm that's quirky, not quite sure why it would do that.  Things I'd be trying:
* Make doubly sure the minidexed.ini I'm changing is the one that is being recognised on the card! (I only say this from experience - always worth a double check by changing something with an alternative obvious effect!)
* Can you test your ssd1306 with some demo code somehow - for me, I'd hook it up to an Arduino to check that its all ok. Alternatively maybe the demo example code from the original circle environment (but this means you'd have to build a kernel image yourself).  You could try an MT-32 Pi image... most of the rest might not work, but it should initialise the display ok...?
* I trust there are no odd messages on the HDMI display that indicates something weird going on?
* Try a different ssd1306 (I have had the odd faulty one from time to time).

That is all that springs to mind right now...?

pegli, 2022-12-28:
Re the suggestions above:

1. I definitely edited the minidexed.ini on the card.  I've already been bitten by that mistake a couple of times. :)
2. Tested with the [SSD1306 display sample](https://github.com/rsta2/circle/tree/master/addon/display/sample/ssd1306) from Circle.  Details below.
3. Nope, no errors on the HDMI display, and both the DAC and the SSD are being detected.
4. Issue was repeatable on three different display units.

I think the issue is the physical connection between the display and the HAT board I built.  I thought that I might have mis-wired the SCK and SDA pins, so I plugged a ribbon cable in between the header and the display, thinking I would use the cable to swap the two pins and see if the problem still occurred.  I used the Circle kernel image for testing.  When the pins were swapped, nothing was displayed, so it looked to me like I had wired the header correctly.  When I ran the cable straight from the display to the header, the Circle kernel ran several times as expected without distortion.  I then booted from my Minidexed SD card and was no longer able to reproduce the distorted display problem, as long as I had the display connected to the board via the ribbon cable.

When I removed the cable and plugged the display straight into the header, the distortion came back...for a while.  I eventually found a position of the display where I can repeatedly boot the Pi and *not* see distortion.  My plan is to yank the header and put in a new one to see if that helps.  The good news is that this doesn't seem to be a software or I2C issue, so I'm closing this ticket.

closed 2022-12-28 by pegli

diyelectromusic, 2022-12-28:
Sounds like you've been able to get some movement on this issue - well done!  How's your electronics?  I'm now wondering about things like:
* I wonder if there is suitable "pull up" on the I2C links... might be worth checking the voltage levels?
* Maybe there is some interference - I can't remember if there are configuration settings used in the code required for the screen that perhaps could be failing, but allowing enough data gets through to show /something/...
* Could the casing be causing problems with an occasional short or something like that?

I think we'd need to see some photos of your physical build to suggest anything else.  But it sounds like you're making some progress now :)

Do let us know if you find out any more (and photos of your build are always welcome regardless!)

Kevin