probonopd/MiniDexed #403: Sustain Pedal NoteOff Issue

Banana71, 2022-12-18:
When you hit a note with high velocity while holding the Susatin Pedal and then pressing the same note a second time with low velocity then cuts off the first sound. That doesn't sound natural. Dexed doesn't behave like that. (And all other keyboard instruments I know)

dcoredump, 2022-12-18:
Synth_Dexed allows two different note refresh modes: one which seems to be activated and works as you described (refresh) and the other which should implement your needs (regtrigger). The magic method is `Synth_Dexed::noteRefreshMode(bool mode)`.

I am currently not sure which is the right assignment, but I think...

1. `mode == false`: Retrigger mode (same note will be triggered again)
2. `mode == true`: Refresh mode (same note will be refresh with new values)

I think adding `m_pTG[i]->setNoteRefreshMode(false);` (or perhaps `true`)

before `minidexed.cpp:99` may solve the problem.

Regards, Holger

donluca, 2023-03-05:
I'll have to check how currently Minidexed behaves, but @Banana71 is correct and that should be the default behavior.

I was thinking that I could add an option in MiniDexed to let the user configure this, but I'd go as far as saying that it should be the default behavior and should be changed in Synth_Dexed directly.

Let me know what you think @probonopd @dcoredump @Banana71 , eventually I can make another PR and hook this up as well.

probonopd, 2023-03-05:
If everyone agrees, then we should make this the default in Synth_Dexed indeed.

Banana71, 2023-03-05:
> If everyone agrees, then we should make this the default in Synth_Dexed indeed.

👍🏼

dcoredump, 2023-03-05:
It's done in [8c67f73d41](https://codeberg.org/dcoredump/Synth_Dexed/commit/8c67f73d413dfbe9830af853bdb173cf1059fd97). Works on MicroDexed as expected.

probonopd, 2023-03-05:
Thanks @dcoredump but now I am getting a compile error; have you seen this before?

donluca, 2023-03-05:
The compile error is unrelated to this and it's due to changes in the engine selection.

dcoredump, 2023-03-06:
> Thanks @dcoredump but now I am getting a compile error; have you seen this before?

Oh, sorry. The AVR compiler hasn't complained... I think I fixed it in [8c67f73d41](https://codeberg.org/dcoredump/Synth_Dexed/commit/a39f556cd3a30595883c08256d3d02b068c1b8af). I installed the toolchain for aarch64 and tried with the common workflow: no error at my side. Can you try again?

donluca, 2023-03-25:
I think this issue can be closed as it now behaves as it should.

Banana71, 2023-03-25:
Uhh, I completely overlooked that. Works great, thanks @dcoredump Holger

closed 2023-03-25 by probonopd