probonopd/MiniDexed #416: Serial MIDI sometimes drops messages when clock is running

JennToo, 2023-01-15:
I have an SQ-64 and noticed that if I send my MIDI keyboard through the SQ-64 in "live" mode, notes will get randomly dropped and stuck. But if I directly connect my keyboard over MIDI, it works just fine.

I enabled MIDI traces and saw that the RPi is getting hammered with active sensing and clocking MIDI messages. If I disable MIDI clocking on the SQ-64 the issue doesn't seem to happen anymore.

However, the sequencer isn't really very useful without a clock, and I haven't found a way to let it run an internal clock without also transmitting it to the MIDI outs.

I'm suspecting the serial buffer gets too full when all the extra spam is coming in, and so random messages start getting lost. I'll do a little more investigating to see if I can diagnose it further.

Also not sure if it makes a difference, but this is an RPi 3 A+.

JennToo, 2023-01-15:
Tracing through things, increasing the serial buffer sizes doesn't help. And it makes sense that it wouldn't, the buffer never seems to get anywhere close to full.

I'm now thinking this might be a state machine issue of some kind with the clock messages being misinterpreted.

probonopd, 2023-01-16:
Thanks @JennToo for looking into this. Do you know a good way to test this, using only PC-side software like REAPER but no special MIDI hardware (which I don't have)?

JennToo, 2023-01-16:
I haven't found a great way, no. I tried capturing through the SQ-64's USB-MIDI interface, but in that case it doesn't seem to intersperse the system real-time messages with the note events. Which also makes me worried that any attempt to try and send this with a USB MIDI device might also fail. I'm wondering if the USB MIDI protocol just packetizes all the events to avoid ever seeing this problem.

diyelectromusic, 2023-01-16:
>  I'm wondering if the USB MIDI protocol just packetizes all the events to avoid ever seeing this problem.

I believe it does, yes, and the rate is so much quicker that these kinds of issues don't really apply in the same way (more here: https://github.com/probonopd/MiniDexed/pull/417#issuecomment-1384468946)

Kevin

PR #417 (2023-01-18): Ignore clock and active sensing on serial MIDI
Fixes #416

I'm not entirely sure this is the correct way to fix it. But it does seem to fix the problem I was seeing. I can now completely mash the keyboard and no notes will get stuck on.
commit 61a301a2 (2023-01-18): Ignore clock and active sensing on serial MIDI (#417)

* Ignore clock and active sensing on serial MIDI

Fixes #416

I'm not entirely sure this is the correct way to fix it. But it does
seem to fix the problem I was seeing. I can now completely mash the
keyboard and no notes will get stuck on.

* Handle all system real time messages in serial
commit 9bc36800 (2023-12-30): Set USB Gadget name to MiniDexed and fix setting of USB product (device) ID.  Fixes #416 (#593)
closed 2023-01-18 by probonopd