probonopd/MiniDexed #421: Latest version of circle has moved the sound drivers

diyelectromusic, 2023-01-18:
The sound drivers in circle have moved to their own directory.  This means updating minidexed.cpp, minidexed.h and Rules.mk to support the new libsound.a library in circle.

diyelectromusic, 2023-01-20:
Looks like there is also an update to circle_stdlib_app.h which we copy from circle-stdlib (I don't know why we copy and not use it directly?).  The circle-stdlib version has changed the call to CGlueStdioInit as per this commit: https://github.com/smuehlst/circle-stdlib/commit/e880dc0ed1a2a3b8da6523383336532797c0be0e

We need to port that over to MiniDexed circle_stdlib.app.h too in order to use the develop version of circle_stdlib.

probonopd, 2023-01-20:
Great point. Added that to the PR, fingers crossed :)

diyelectromusic, 2023-01-20:
I think we've both done it - never mind!  The commit history is going to get very "noisy"!! :)

probonopd, 2023-01-20:
> I think we've both done it

But yours is working - big difference ;-) :+1:

diyelectromusic, 2023-01-22:
This update has been incorporated into https://github.com/probonopd/MiniDexed/pull/422 so I'm closing this issue.

closed 2023-01-22 by diyelectromusic