probonopd/MiniDexed #458: Bank menu has blank entries for unloaded banks

diyelectromusic, 2023-03-31:
From: _Originally posted by @donluca in https://github.com/probonopd/MiniDexed/issues/455#issuecomment-1484175202_

Blank entries, gaps, or invalid banks should be skipped in the UI but at present it shows blank entries.

donluca, 2023-03-31:
I'm working already on automatically skipping empty voices.
I think I can use the same approach (read the bank/voice name) for empty banks.

diyelectromusic, 2023-03-31:
Ok, for ease of integration I would recommend that each fix should have its own PR - this was one of the problems with some of the outstanding PRs in the list - there were a lot of changes in each PR making it hard to integrate them.

What else are you working on, so I'm not wasting my time :)

Kevin

donluca, 2023-03-31:
Yeah, that's what I've been doing if you look at my PRs: each PR has a single "fix" or improvement.

To be honest I'm making some changes mainly for myself: I've merged #395 so that I can scroll beyond 127 banks and I've made a change in the UI so that in the voice selection you can see the name of the bank at the top (trimmed if longer than 16 characters).
I'm not sure people would agree with those kind of changes, so I've made a [private branch](https://github.com/donluca/MiniDexed/commits/private) for myself and will create separate branches for things I'd like to see merged into MiniDexed.

Right now I'll just be tackling the empty voices thing, if you want to take care of the empty banks, the floor is yours.

diyelectromusic, 2023-03-31:
Ok :)
In that case I'll kick off a series of commits in my own branch for some of these minor fixes without changing too much at a time, so hopefully we can converge on something useful!  Assuming @probonopd agrees of course.

diyelectromusic, 2023-03-31:
Ok, so Bank Select for > 127 banks (via menu and MIDI) is now in this branch: https://github.com/diyelectromusic/MiniDexed/tree/bankhandling

I don't suppose you fancy giving it a go?

I'll get going on some of the other changes if that all seems ok.

Kevin

donluca, 2023-04-01:
I'll give it a go this evening if I have a bit of time.

It would greatly help if you could provide a build (just kernel7.img suffices, as I use a RPi2), unless you want me to try and merge this into my private branch... ?

probonopd, 2023-04-01:
Build for testing:
[MiniDexed_2023-03-31-36097b0](https://github.com/probonopd/MiniDexed/suites/11955082990/artifacts/626889921)

donluca, 2023-04-01:
Can confirm this is working as expected.

closed 2023-04-01 by probonopd