probonopd/MiniDexed #467: MiniDexed should return to first voice when changing banks

donluca, 2023-04-03:
I don't own a real DX7 so I have no idea of how it behaves, but in all the synths I know, when you change a bank you get sent to the first voice of that bank.

In MiniDexed, if you are at, for example, voice 22 of one bank and then change to another bank, it will take you to the 22nd voice of that bank.

From my experience, this is not correct and when changing banks you should find yourself at the first voice of chosen bank.

I already fixed this in my own private branch, let me know if this is the correct behavior you'd expect on DX7/Minidexed and I'll do a PR.

diyelectromusic, 2023-04-03:
This was discussed a fair bit in #425 but not really resolved.  Some DX synths switch to the current patch number in the new bank but the bank/program handling is currently pretty much the same for MIDI and the menus, and the MIDI spec says:
_"When the Bank Select message is received, the receiving device must remember that bank number in readiness for the following Program Change. Bank Select alone must not change the program. This is to assure that multiple devices change 
concurrently."_

So it isn't clear quite what should be done.  Basically the bank selection menus are pretty much a pseudo "send MIDI bank select" message rather than a real "select the next bank" menu.

But I agree, I think I'd expect the bank change to switch to a voice in the new bank.  But which one - the first or the "current" number?

Kevin

Banana71, 2023-04-03:
With the TX816, the banks are structured in such a way that the voice numbers are adapted to the voices. 

 I would stay by the current voicenumber.

donluca, 2023-04-03:
> Update: if you watch the video you can see that the original DX7 does not change program when changing banks.

Ok, so the current behavior is the "expected" one.

If there are no further observations/discussion we can close this.

probonopd, 2023-04-03:
Thanks @donluca