probonopd/MiniDexed #470: Number of performance files needs bounds checking

diyelectromusic, 2023-04-04:
There is no bounds checking on the number of performance.ini files that will be read in during CPerformanceConfig::CreateNewPerformanceFile(void).  There is a hard-coded limit of 1024 set for m_nPerformanceFileName in performanceconfig.ini

Kevin

diyelectromusic, 2023-05-12:
I was going to fix this, but testing it is a nightmare... do we really want 1024 as the limit for performances?  That will probably take an age to load off the card and will certainly take an age to scroll through via the UI! :)

I'm wondering if a few hundred is a more realistic limit?

But happy to stick with 1024 if that is really what people want, but someone else will have to test it :)

Paging @probonopd, @Banana71 and @donluca mostly I guess, but all views welcome!

Kevin

Banana71, 2023-05-12:
my thoughts on it:

The limit of the maximum number of performances was previously 128. It doesn't have to be 1024 performances.  I would like a folder structure: e.g.  4 folders with 64 or 128 Performances each.  So you could copy and sort the sounds for different applications individually in the folder.

Currently there are about 100 performances on my SD card.  A quick sound selection is not possible.  It would also be helpful if the performance number is shown in the top left of the display.  The number could be displayed inverted for loaded Performances.

diyelectromusic, 2023-05-12:
I wasn't planning on re-writing or updating the implementation at this stage, I just want to stop things like [this ](https://github.com/probonopd/MiniDexed/issues/379) happening again.  Crashes due to built-in limits with no checking really aren't nice :)

Although if someone is dedicated enough to get 1024+ performances on their device, then maybe we should all just be stepping back and applaud the resulting crash!

Kevin

Banana71, 2023-05-12:
If I remember correctly the problem wasn't the number of performances but the numbering of the last performance.  If the limit is exceeded, it crashes when saving a new performance.  As we now select and load Performances, a limit of 256 is more than sufficient.

diyelectromusic, 2023-05-13:
Ok. If @probonopd agrees, then I'll update it to 256, attempt to test it properly and send that one in too :)

Kevin

PR #494 (2023-05-14): Issue470 fix - Performance file bounds checking and error handling
Introduces bounds checking for performances; reduces number of performances to 256; includes better error handling in the UI and on the display.

closes #470
closed 2023-05-14 by probonopd