probonopd/MiniDexed #471: TGs in OMNI mode should ignore AllNotesOff messages

diyelectromusic, 2023-04-04:
As commented here: https://github.com/probonopd/MiniDexed/pull/366#issuecomment-1334330697 regardless of any config settings if a TG is listening in OMNI mode then it should not respond to All notes Off MIDI messages (more here: https://github.com/probonopd/MiniDexed/issues/356#issuecomment-1261319623)

The MIDI Spec ("ALL NOTES OFF" section) says this about All Notes Off:
_"All Notes Off should only turn off those notes that were turned on via MIDI. If an instrument cannot differentiate between its local keyboard and incoming MIDI messages, All Notes Off should be ignored."_

As MiniDexed has no keyboard and only responds to MIDI this is ok.

_"Receivers should ignore an All Notes Off message while Omni is on (Modes 1 & 2)."_

There are other comments about interaction with some of the other channel mode messages too and the "Additional Explanations and Application Notes" appendix has even more to say on interaction with modes, use in MIDI merging, interaction with a HOLD pedal but I don't think we need to worry too much at this level...

Kevin

closed 2023-05-13 by diyelectromusic

probonopd, 2023-05-13:
Pro tip: If you write "closes #417" in the commit message (instead of "Fix for Issue #471"), then the ticket is closed automatically at the time when the PR is merged.

diyelectromusic, 2023-05-13:
> Pro tip: If you write "closes #417" in the commit message (instead of "Fix for Issue #471"), then the ticket is closed automatically at the time when the PR is merged.

Ok thanks will do if that is how you'd like things to work.

I'd not considered any kind of auto-closing as I guess I was taking the view that an issue is only known to have been fixed once the PR has been confirmed to have been successful by someone else rather than just when it was merged, but happy to go either way!

Kevin