probonopd/MiniDexed #497: classic midi to minidexed

Synthfreek, 2023-05-24:
Dear friends, please help. I try to built a midi in for the raspberry pi 3b that is running minidexed (the dx 7's)
I want to use a midikeyboard that has only the classic 5-pin midi
I did built a midi in like descripted on https://diyelectromusic.files.wordpress.com/2021/02/diy-midi-receive-3.3v-v2_schem-1.png but it seems not to work. do I have to change something in the ini code? 
any suggestions are welcome
thanx, kind regards, Wim (the Netherlands)

diyelectromusic, 2023-05-25:
MIDI should "just work", but there is a setting you can enable to make sure MIDI data is getting through. Set MIDIDumpEnabled=1 in the config file and ensure there is a video output (e.g. HDMI) connected (not an LCD).

Then it is a case of check and double check the circuit really.  Can you provide a hi-res photo so we could have a look to see if we can spot anything?

The thing that almost always gets me - even now, and I've built more MIDI circuits than I can remember - is the pinouts for the DIN sockets.  I still get 4 and 5 wrong!

Kevin

Banana71, 2023-05-25:
Some good photos of your circuit would help 👀

Synthfreek, 2023-06-10:
[image]
[image]
[image]

diyelectromusic, 2023-06-10:
Well I'm not sure I can make out which pin is which on that DIN Connector!  I just have to assume you've checked and double checked.  The rest of the circuit looks ok, (I trust white is connected to GND - its a little hard to see).  So assuming nothing weird is happening on the solderless breadboard (they are quite unreliable at times I'm afraid), then all I can suggest is enabling MIDI debugging and starting to look at the signal at different places as I suggested in my previous comments https://github.com/probonopd/MiniDexed/discussions/279#discussioncomment-6032166

Kevin

Gustavomurta, 2023-07-20:
Hi Synthfreek,
Analyzing the photo of the MIDI RCA-5 female connector, confirm that the blue wire is pin 5 and the red wire is pin 4. See if you can see the pin numbering on the connector.

[image]

Gustavomurta, 2023-07-20:
Another diagram made by me.

[image]

Synthfreek, 2023-07-22:
I will try the other way around. The picture may look confusing because I bended the unused pins to the outside.

closed 2024-02-01 by probonopd