probonopd/MiniDexed #502: SysEx Issues: Master Volume and Device ID

diyelectromusic, 2023-05-28:
The master volume SysEx System Common message is only valid if used with the special manufacturers ID of 0x7F, so this should be included in the test in mididevice.cpp (https://github.com/probonopd/MiniDexed/blob/main/src/mididevice.cpp#L178).

Also, in the TG specific SysEx handling later on, there is no confirmation that the SysEx message is from Yamaha, although this happens within Synth_Dexed itself so is probably ok.

I am also now wondering about the device ID in the SysEx messages.

The device ID is assumed to be a MIDI channel, so is checked against the MIDI channel for the TG.  This is probably an assumption as Device IDs are meant to be something different I think?  I don't know how the TX816 uses them, but it might make more sense to check the Device ID against the TG number itself rather than MIDI channel?  But it probably depends on how a DX editor would use Device ID (Dexed itself only seems to have the concept of "sysex channel")?

The DX7II Advanced MIDI Application note says this about device IDs:
[image]

I've not found any mention of them for the original DX7 so far (did that even have them in the UI?).

Device IDs also appear in the Universal System Exclusive Messages (0x7F messages) too and it seems other manufacturers often have a device ID after the manufacturer's ID too, but I guess that isn't mandatory.  Here is what the MIDI spec says about Universal SysEx Device IDs:
[image]

So is assuming it is the same as MIDI channel the simplest way to go for someone using MiniDexed?  Maybe it is... but we perhaps ought to have a note about it somewhere and it might make exchanging data with a real DX7II problematic if the UI on the real synth is used to set Device ID to something else (does it default to 1?)

Kevin

probonopd, 2023-05-29:
Hi @Kevin, did you run into any concrete issue with the current implementation, like something misbehaving?

diyelectromusic, 2023-05-29:
No, I was just working through the code for other reasons and noticed that this is possibly wrong.  It is the kind of problem that will only come up in specific circumstances, but if it does it will quite a pain to spot, hence documenting it here to see what others think.

The issue with the Master Volume is that we might interpret a different SysEx message as Master Volume.

On Device ID, I need someone with a real DX7 or DX7II or TX816 to tell us what the Device ID is meant to be and if this is ok! :)

Kevin

PR #775 (2024-12-14): Fix Device Master Volume Control handling
And also comment out low-level MIDI dump code that really should only be there if someone is messing around with the MIDI handler itself.
-- actually no, that is already commented out in the serial driver.  But this is very unresponsive due to line-buffer flushing, etc, so switched to fprintf to stderr instead.

This is a fix for Issue #773 and partly fixes #502 - at least the Master Volume Control part - the message is now explicitly checking all SysEx bytes and the message length to ensure it really is a MIDI SysEx realtime device control message for [... cut]
closed 2024-12-14 by probonopd