probonopd/MiniDexed #503: MiniDexed_2023-05-14-87dc5ce doesn't boot on Pi 3 B V1.2

Andy2No, 2023-05-31:
This is my first attempt at getting MiniDexed to work, and my first time using a Raspberry Pi, but this build doesn't appear to be able to boot, on mine.

The last messages were:

00:00:02.95 Performance: Number of Performances: 88
00:00:02.95 mcore: CPU core 1 started
00:00:02.95 mcore: CPU core 2 started
00:00:02.96 mcore: CPU core 3 started

It then hangs forever.

I'll try earlier builds.

Andy2No, 2023-05-31:
MiniDexed_2022-08-25-dcbd04c gives the same result, stopping at the same message.

Andy2No, 2023-05-31:
MiniDexed_2022-03-11 boots but gives a long stream of messages about chunksize, and apparently refuses to accept my USB-MIDI device, which is a blue plastic M-Audio MIDISport Uno 1X1, with Midiman on it.

The messages are:

Getchunk: maximum duration was 56us (2%)
(over and over)

and a short series of messages which is a little hard for me to write down before it scrolls out of view.  The gist was device found, with a vendor ID, device has no supported function (which I've read isn't an error??) then device removed.

There's no sound, yet, and I have no experience of this working, so I have no way of knowing what's considered normal.  FWIW, scrolling error messages are not helpful, when having to write things down off a screen.  Since it's the same one over and over (Chunksize), perhaps it could just have a count at the end of the line, and not scroll up?

Is there any way of seeing a version history for the releases, with the Pi and USB devices that had been added, or perhaps bug fixed in that release?

Andy2No, 2023-05-31:
Still on MiniDexed_2022-03-11, I've tried a different blue plastic USB-MIDI device, this time an M-Audio Uno (variation on the same theme as the other one.

This one is accepted, and I can get sound from the 3.5mm audio jack socket, sending notes to the USB-MIDI device, via another one connected to my PC.

I still have the scrolling Chunksize messages on the HDMI display.  Whatever the reason for that is, it would seem to be taking up processor time to send those messages, which might be better used doing something else.  So far, I've failed to find any information about that.

Anyway, the fact that I can get it to respond to MIDI notes, and also Program Change, using this build from March 2022, shows that it can basically work, but not with the two most recent builds on the Releases page.

Banana71, 2023-05-31:
You have a Raspberry Pi Model 3 B, PWM output, Midi Controller M-Audio, no display, no rotary encoder, no DAC.  
There are problems with some M-Audio controllers.  With your simple configuration, every miniDexed version should run.  Is your SD card formatted correctly?

[image]

Andy2No, 2023-05-31:
> Is your SD card formatted correctly?

I formatted it with the SD Association formatting tool, using the latest version on the default settings, which is FAT32, with 32k blocks, IIRC.

It now appears to be working fine, on this older build, other than the stream of chunksize messages, which I don't know enough about to know whether that's normal.  It wasn't even able to boot, on the two most recent builds in Releases.

diyelectromusic, 2023-05-31:
Can you try booting it up with no USB device attached and post a photo of the boot screen messages?

Also, we'd need to know your exact hardware setup and configuration if we're to get anywhere trying to figure out what is going on.

> It then hangs forever.

How do you know it hangs?  Has the UI/controls become unresponsive? What UI/controls are you using?

There are typically no boot messages after this point so I'd only expect something on the screen if something was going wrong...

Try running it with this in the minidexed.ini file:
MIDIDumpEnabled=1

and see if you get MIDI events logged to the display.  What MIDI device where you using for your first attempt?  USB or serial?  What is the MIDI circuit if serial?  What MIDI channel settings are you using for the Tone Generators?

> Whatever the reason for that is, it would seem to be taking up processor time to send those messages, which might be better used doing something else. So far, I've failed to find any information about that.

The HDMI display is just for diagnostic messages, it isn't a user interface for the device.  Yes it takes up processing time, but there is something going wrong on your system but without this you'd never know!?

Basically, we really need a little more to go on, but I'd strongly recommend going back to the latest release and then we can start from a known good point, but detail exactly how it isn't working and how your system is set up.

Kevin

diyelectromusic, 2023-05-31:
> Getchunk: maximum duration was 56us (2%) (over and over)

From what I can see this message is a diagnostic aid that should only be seen if the following is enabled in the config file:
ProfileEnabled=1

So in a more fuller response to "which might be better used doing something else" this is a debug setting and wouldn't normally be happening anyway.  I don't know why that might be enabled, unless you've managed to find an older build from the earlier development phases where this was on by default - i.e. before it became a config setting...

As I say, please revert to the latest build, enable MIDI diagnostics (and yes, this will be more processor time taken to try to solve your problem by printing to the screen), and let us know the symptoms of your system along with details of your hardware build and minidexed.ini and performance.ini configuration and if you have any special settings in either config.txt or cmdline.txt.

Kevin

Andy2No, 2023-05-31:
@diyelectromusic I tried the last two releases with just the Raspberry Pi 3 B V1.2 and a HDMI cable, and the same thing happened - it hanged after the core 3 message, and never got any further.

I'll get to the screen photos shortly, but all I did was try the exact files from the releases, on a new 32GB microSD card formatted as above, in the Pi 3 B, with nothing attached other than a HDMI cable to my TV/monitor, so I could see the messages.

The success I've had with the older release, MiniDexed_2022-03-11, is tempered by a lot of stuck notes and some missed notes, but I guess that's a separate issue from the subject of this thread.

diyelectromusic, 2023-05-31:
> @diyelectromusic I tried the last two releases with just the Raspberry Pi 3 B V1.2 and a HDMI cable, and the same thing happened - it hanged after the "mcore: CPU core 3 started" message, and never got any further.

As I say, when working I'd expect nothing else on the display after that, so once again, how did you conclude it is hanging especially if you have no user interface...?  Could there be an issue with the MIDI configuration for example?

Try MIDIDumpEnable=1 in minidexed.ini - do you have a USB MIDI keyboard controller you could try with it?

Kevin

diyelectromusic, 2023-05-31:
For what it's worth, I've just downloaded the same ZIP release, expanded all files onto a new SD card, plugged it into my Pi 3B, plugged in audio to PWM, powered it up with a HDMI monitor, and plugged in a USB MIDI keyboard controller and it just plays and makes sounds.

The only thing on the screen is the additional USB messages showing the plugging in of the USB keyboard.

Kevin

Andy2No, 2023-05-31:
I concluded it didn't work because there was no response from trying it with my original USB-MIDI device, the MidiSport 1x1, and the fact that when I looked at the boot messages it had the appearance of having hanged/hung, after showing that last message.

I'm used to seeing boot messages that give some sense of completion, or even triumph, over the boot process.

I do have a user interface, I just didn't go into all the details, since it didn't seem relevant.  I mentioned that I tried sending notes from my PC, via another USB-MIDI interface semi permanently attached to the PC, after getting it working with older build and the USB-MIDI device that did work (M-Audio Uno).  It didn't seem relevant that I then tried it with the USB-MIDI device plugged into a keyboard (Yamaha CBX-K1, with DIN MIDI), which then gave me lots of stuck notes, and some missed notes, because that seemed outside the scope of this issue thread.

So, to get back to the screenshots, here are three, with the latest release and no changes to the files.  The first one shows it booting with no USB-MIDI device, the second shows it having booted then having the M-Audio Uno USB-MIDI device plugged in afterwards.  The third shows it booted with the M-Audio Uno plugged in already.  

So far, so good.  I'll try it with the keyboard again, next.  I do have other keyboards I can try, so I won't immediately rush to post about that if it works but I get stuck notes with that one.

I actually don't find it noisy.  Noise on USB devices is usually a power supply issue, so I guess I have an unusually quiet one.  Often, the noise goes away if you try running off a rechargeable power brick, instead of a wall wart. 

 I do have a couple of DACs I could wire up (in small module form, not as hats), and I intended to, originally, but unless that helps performance in some other way, I might not bother, because it will draw more power.

[image]

[image]

[image]

probonopd, 2023-07-04:
Closing. The screenshots show that it has booted successfully and all CPU cores have been started. 

Please open a new issue in case your MIDI controller(s) are not recognized.

closed 2023-07-04 by probonopd