probonopd/MiniDexed #511: Loading patches over sysex is corrupted about 35% of the time, 0.96

SeanLuke, 2023-06-22:
I maintain a large patch editor toolkit, Edisyn, which has a DX7 editor.  Up until 0.9.4 (on Intel or M1) Dexed worked perfectly with sysex sent it via Edisyn over a Mac MIDI Loopback.  I upgraded to 0.9.6, which likely is running M1 native, and about 35% of the time sysex patches sent it will arrive corrupted, producing some nasty BZZZZ sound rather than the actual patch.  If you resend the patch once or twice, Dexed recognizes it and plays it properly.   Downgrading back to 0.9.4 and Dexed works flawlessly again.

I am pretty certain that this bug is not Edisyn: the patch sysex appears correct.  I also know that it is not due to timing issues (such as pausing a long time after a patch is dumped before playing a note, etc.).  It appears to be a Dexed error.  I do not know if this bug is due to 0.9.6 or due to the change in Dexed from running Intel (or on Rosetta) vs. running natively on the M1.   But it is definitely a showstopper for patch editors using Dexed.

diyelectromusic, 2023-06-23:
Is this an issue with Dexed or MiniDexed?  If I'm reading your context directly, I believe you might be talking about Dexed in which you'd have to report this against Dexed itself here: https://github.com/asb2m10/dexed

This repository is for MiniDexed a version of Dexed that runs on "bare metal" Raspberry Pi only (i.e. embedded, with no operating system).

Kevin

SeanLuke, 2023-06-23:
I am a moron.  Moving to Dexed.

closed 2023-06-23 by SeanLuke

diyelectromusic, 2023-06-23:
Haha!  No worries!  Good luck :)