probonopd/MiniDexed #526: LFO Delay and LFO Key Sync no funktion

Banana71, 2023-08-05:
No matter what value you set for the LFO Delay (TG1 -TG8), the LFO starts working immediately without delay. The same goes for LFO Key Sync. 

_

> LFO Delay Introduces a delay between the time at which a key is pressed and the time at which the Low Frequency Oscillator (LFO) modulation begins. Can be set from 0 (no delay, modulation begins instantly) and 99 (longest delay)

_

Banana71, 2023-08-07:
To clarify, I created a test sound. The first lower note is from Dexed and there is a noticeable delay before the LFO kicks in. The higher tone comes from the miniDexed and there is no noticeable delay. The tones repeat once. 
The first note (Dexed) plays as expected and the second note (miniDexed) is missing the delay.

https://github.com/probonopd/MiniDexed/assets/104296205/e9c15631-0297-4a52-b6ef-ca54feade02f

probonopd, 2023-08-07:
It'd be interesting to know how MicroDexed behaves in this regard, maybe @dcoredump can give it a try?

dcoredump, 2023-08-08:
Thanks for reporting! I will check this as soon as I can find some free time and come back.

dcoredump, 2023-08-08:
Ok, I replaced my code in Synth_Dexed with the original code for LFO handling. Maybe I have understand something wrong before. Can you check [c9f52741a802ad9bb01263823650f7cc3b0b5108](https://codeberg.org/dcoredump/Synth_Dexed/commit/c9f52741a802ad9bb01263823650f7cc3b0b5108)?

probonopd, 2023-08-08:
Hi @dcoredump, thanks for your quick response. I am running into a compile issue:
https://github.com/probonopd/MiniDexed/actions/runs/5800408137/job/15722456911#step:5:2815
Have you seen this before and do you know the fix? Do we need a newer version of some (which?) component?

Banana71, 2023-08-08:
Is the bug found in #403 also fixed then?

probonopd, 2023-08-08:
> Hi @dcoredump, thanks for your quick response. I am running into a compile issue:
https://github.com/probonopd/MiniDexed/actions/runs/5800408137/job/15722456911#step:5:2815
Have you seen this before and do you know the fix? Do we need a newer version of some (which?) component?

Possibly for all new Synth_Dexed versions we need to merge https://github.com/probonopd/MiniDexed/pull/443 first (which I hadn't merged yet until we know why there is increased CPU usage).

probonopd, 2023-08-08:
Please try this build:
[MiniDexed_2023-08-08-e4ad4cd](https://github.com/probonopd/MiniDexed/suites/14929214521/artifacts/851016206)

It's from #445 with Synth_Dexed updated to [c9f52741a802ad9bb01263823650f7cc3b0b5108](https://codeberg.org/dcoredump/Synth_Dexed/commit/c9f52741a802ad9bb01263823650f7cc3b0b5108)

Banana71, 2023-08-09:
Great joy, the LFO now works as expected and the sustain pedal also holds double played notes. I am now checking all performances and will adjust the LFO if necessary. Thanks to both of you @probonopd & @dcoredump

closed 2023-08-10 by Banana71