probonopd/MiniDexed #534: Breakpoint is displayed an octave too high

Banana71, 2023-08-14:
In Dexed I can set the values from A-1 - C8. 
The values **_A1 - C9_** are possible in the miniDexed.(wrong)

The wiki is also wrong:
Break Point Keyboard Level Scaling allows to raise or lower the Envelope Generator (EG) levels for keys to the left and right of any key specified as the Break Point key. Can be set from _**A1 to C8**_

Correct would be:
Break Point Keyboard Level Scaling allows to raise or lower the Envelope Generator (EG) levels for keys to the left and right of any key specified as the Break Point key. Can be set from _**A-1 to C8**_

None of this has any effect on the sounds, they are played back correctly. Only the display in the miniDexd for the breakpoints is shifted up one octave

probonopd, 2023-08-14:
Thanks @Banana71 for noticing this.

* A-1 is MIDI note number 21
* Apparently there is no A0. Unlike computer scientists, musicians don't seem to like the number 0?
* A1 is MIDI note number 33
* So do we need to subtract 12 somewhere?

This is what we are doing:

https://github.com/probonopd/MiniDexed/blob/b2c95bda78ba7625b7d290813f3db4e4fb64c110/src/uimenu.cpp#L1073-L1078

Looks like we don't even know notes lower than A1 at the moment:

https://github.com/probonopd/MiniDexed/blob/b2c95bda78ba7625b7d290813f3db4e4fb64c110/src/uimenu.cpp#L310-L321

The related Synth_Dexed functionality seems to be:

https://codeberg.org/dcoredump/Synth_Dexed/src/commit/c9f52741a802ad9bb01263823650f7cc3b0b5108/src/dexed.cpp#L1328-L1342

Not 100% sure whether this is the correct fix in `uimenu.cpp/.h`:

```
const char CUIMenu::s_NoteName[112][5] =
{
	"A-1", "A#-1", "B-1", "C-1", "C#-1", "D-1", "D#-1", "E-1", "F-1", "F#-1", "G-1", "G#-1",
	"A1", "A#1", "B1", "C1", "C#1", "D1", "D#1", "E1", "F1", "F#1",	"G1", "G#1",
	...
```

probonopd, 2023-08-14:
Please see #536 for a build for testing @Banana71. Thank you.

diyelectromusic, 2023-08-14:
> * A-1 is MIDI note number 21
> * Apparently there is no A0. Unlike computer scientists, musicians don't seem to like the number 0?

Hang on - that's not true :)  A0 is MIDI note number 21... See https://en.wikipedia.org/wiki/Scientific_pitch_notation

Kevin

probonopd, 2023-08-14:
@Banana71 said an octave too high though, not two octaves... hence I am puzzled...
In any case, building a version with two additional octaves now.

diyelectromusic, 2023-08-14:
Ok, so on this topic, this is an extra from "The Complete DX7" chapter on Keyboard Scaling:

[image]

[image]

The keyboard itself is described as being C1 to C6.

On setting the breakpoint:

[image]

However in terms of actual voice data, the MIDI Technical Data and Charts shows:

[image]

So if 0 is A-1, then 99 takes us up to just past A7 at 96 I think, up to 99  = C8 I think?  Note, these are not MIDI numbers, they are DX7 0..99 parameter numbers.  In MIDI note numbers, 0 is C-1 not A-1...

Kevin

Banana71, 2023-08-14:
I haven't tested the version yet but... 
Dexed shows me the following values for the breakpoints:
I have taken the following values from Dexed and put them into the format.
**In Performance.ini is C3 = 60 for NoteLimit, mayby thats another Format.**

const char CUIMenu::s_NoteName[112][5] =
{
"A-1", "A#-1", "B-1", "C0", "C#0", "D0", "D#0", "E0", "F0", "F#0", "G0", "G#0",
"A0", "A#0", "B0", "C1", "C#1", "D1", "D#1", "E1", "F1", "F#1", "G1", "G#1",
"A1", "A#1", "B1", "C2", "C#2", "D2", "D#2", "E2", "F2", "F#2", "G2", "G#2",
"A2", "A#2", "B2", "C3", "C#3", "D3", "D#3", "E3", "F3", "F#3", "G3", "G#3",
"A3", "A#3", "B3", "C4", "C#4", "D4", "D#4", "E4", "F4", "F#4", "G4", "G#4",
"A4", "A#4", "B4", "C5", "C#5", "D5", "D#5", "E5", "F5", "F#5", "G5", "G#5",
"A5", "A#5", "B5", "C6", "C#6", "D6", "D#6", "E6", "F6", "F#6", "G6", "G#6",
"A6", "A#6", "B6", "C7", "C#7", "D7", "D#7", "E7", "F7", "F#7", "G7", "G#7",
"A7", "A#7", "B7", "C8"
};
static const unsigned NoteC3 = ?

This is from your code:
const char CUIMenu::s_NoteName[112][5] =
{
**"A-1", "A#-1", "B-1", "C-1", "C#-1", "D-1", "D#-1", "E-1", "F-1", "F#-1", "G-1", "G#-1",
"A1", "A#1", "B1", "C1", "C#1", "D1", "D#1", "E1", "F1", "F#1", "G1", "G#1",
"A2", "A#2", "B2", "C2", "C#2", "D2", "D#2", "E2", "F2", "F#2", "G2", "G#2",
"A3", "A#3", "B3", "C3", "C#3", "D3", "D#3", "E3", "F3", "F#3", "G3", "G#3",
"A4", "A#4", "B4", "C4", "C#4", "D4", "D#4", "E4", "F4", "F#4", "G4", "G#4",
"A5", "A#5", "B5", "C5", "C#5", "D5", "D#5", "E5", "F5", "F#5", "G5", "G#5",
"A6", "A#6", "B6", "C6", "C#6", "D6", "D#6", "E6", "F6", "F#6", "G6", "G#6",
"A7", "A#7", "B7", "C7", "C#7", "D7", "D#7", "E7", "F7", "F#7", "G7", "G#7",
"A8", "A#8", "B8", "C8", "C#8", "D8", "D#8", "E8", "F8", "F#8", "G8", "G#8",
"A9", "A#9", "B9", "C9"**
};
static const unsigned NoteC3 = 39;

diyelectromusic, 2023-08-14:
As I said in the other conversation, in Musical terms, notes go A1, A#1, B1, C2, C#2... not A2, A#2, B2, C2... :)

Banana71, 2023-08-14:
Then these values should actually be the right ones!?

"A-1", "A#-1", "B-1", "C0", "C#0", "D0", "D#0", "E0", "F0", "F#0", "G0", "G#0",
"A0", "A#0", "B0", "C1", "C#1", "D1", "D#1", "E1", "F1", "F#1", "G1", "G#1",
"A1", "A#1", "B1", "C2", "C#2", "D2", "D#2", "E2", "F2", "F#2", "G2", "G#2",
"A2", "A#2", "B2", "C3", "C#3", "D3", "D#3", "E3", "F3", "F#3", "G3", "G#3",
"A3", "A#3", "B3", "C4", "C#4", "D4", "D#4", "E4", "F4", "F#4", "G4", "G#4",
"A4", "A#4", "B4", "C5", "C#5", "D5", "D#5", "E5", "F5", "F#5", "G5", "G#5",
"A5", "A#5", "B5", "C6", "C#6", "D6", "D#6", "E6", "F6", "F#6", "G6", "G#6",
"A6", "A#6", "B6", "C7", "C#7", "D7", "D#7", "E7", "F7", "F#7", "G7", "G#7",
"A7", "A#7", "B7", "C8"
};
static const unsigned NoteC3 = 39

Banana71, 2023-08-14:
I'm totally confused now. 🥺
@probonopd did you take the value 60 for C3?
would   static const unsigned NoteC3 = 39 be correct?

diyelectromusic, 2023-08-14:
> I'm totally confused now. 🥺 @probonopd did you take the value 60 for C3? would static const unsigned NoteC3 = 39 be correct?

Yes, I think it should be 39 (see my comments here: https://github.com/probonopd/MiniDexed/pull/536#issuecomment-1677911343)

Kevin

diyelectromusic, 2023-08-14:
Also, there are now 100 elements in that array again, not 112...

Kevin

Banana71, 2023-08-14:
> Also, there are now 100 elements in that array again, not 112...
> 
> Kevin

I only took over the values that are displayed in the Dexed UI.

Banana71, 2023-08-15:
Thank you so much @probonopd and @diyelectromusic 

https://github.com/probonopd/MiniDexed/pull/536#issuecomment-1678479411

closed 2023-08-15 by Banana71