probonopd/MiniDexed #535: Initialize display earlier

probonopd, 2023-08-14:
With many performances and voices on the SD card, it can take a while before anything gets displayed on the display. We should print something during this time, e.g.,

```
* MiniDexed    *
* Synthesizer  *
```

so that users know the device is "alive".

For this, we may need to move the display initialization earlier in the boot process, before voices or performances get loaded from the SD card.

patrickdowling, 2023-08-18:
Heh, that's pretty much what I came to after poking around.
Just I'm apparently not the target audience (yet) -- it just blips quickly during startup :)
Which maybe also invalidates the idea of adding  version number/build tag there.

probonopd, 2023-08-18:
Tested the PR with a big set of voices and performances loaded, it makes the waiting time more bearable. Thanks @diyelectromusic