probonopd/MiniDexed #539: TG1 - TG8 Volume Bar graph from 0 - 14

Banana71, 2023-08-15:
Why do we have a bar graph for adjusting the volume? 
Internally the miniDexed works with a resolution of 0 - 99 but we use a coarse resolution of 0 - 14 (scaled to 0 - 99). A finer adjustment via MIDI is possible with a controller keyboard, but then all TG's are adjusted on the same MIDI channel.
A finer resolution would be desirable, especially when using the miniDexed as a multitimbral module. It should be possible to change the volume parameter to 0-99.

What do you mean?

Banana71, 2023-08-15:
> Do we really want to have to press the button 99 times?

No, that's why I have an encoder :-)
But I think it's important to be able to adjust the parameters over the entire range.
With most electronic devices with switch operation, you would hold down the button and then parameters would be adjusted more quickly after a short time. e.g.

probonopd, 2023-08-15:
Good idea.

diyelectromusic, 2023-08-17:
I think the volume is adjustable over the entire range - see https://github.com/probonopd/MiniDexed/blob/main/src/minidexed.cpp#L494

But just the display is quantised to blocks as the default display is "text only"... see https://github.com/probonopd/MiniDexed/blob/main/src/uimenu.cpp#L1006

Kevin

probonopd, 2023-08-17:
That's actually how I _hoped_ it would be :)

probonopd, 2023-08-17:
If we have full granularity via MIDI CC then I think it's fine as-is.

closed 2024-11-15 by probonopd