probonopd/MiniDexed #541: MIDIAutoVoiceDumpOnPC and Performances

Banana71, 2023-08-18:
I have arrived in modern times and edit the voices in the miniDexed with Dexed via MIDI. Changes on the PC in Dexed are adopted in miniDexed. So far so good. :-)

What I miss the most about miniDexed is the ability to save Voices. I usually fine-tune performances directly in the miniDexed. These Voice changes cannot be saved as a Patch, only in the Performance.

I hoped that the MIDIAutoVoiceDumpOnPC=1 function would allow me to transfer these patches from the performances to the PC. (reverse engineering) . When loading a performance, the sysex data of all TG's are transmitted, regardless of which MIDI channel is set in the performance (1-16, off). Unfortunately, the data of the VoiceData stored in the Performance is not transferred, but the data of the Voice linked to the Performance on the SD card (BankNumber, VoiceNumber of the folder /sysex/voice/...) is sent to Dexed .

I can copy the voices in sysex files on the SD card to the PC at any time and then edit them via Dexed. What is important are the voices stored in the performance.
This is all very complicated to describe and I hope you understand what I mean.
Am I doing something wrong? Or the function can be changed?
Can it be changed so that the VoiceData(s) are sent to the PC (Dexed) on the corresponding MIDI channel. The Voice Data from TG's that have the MIDI channel set to off should not transmit any data.

https://github.com/probonopd/MiniDexed/issues/386#issue-1486829987  I read but didn't find out

----
e.g.

`BankNumber1=1`
`VoiceNumber1=9`
MIDIChannel1=1
Volume1=68
Pan1=0
.
.
.
`VoiceData1=3E 0B 0E 0C 5C 63 5C 00 24 04 03 00 00 01 00 01 50 00 01 00 0A 32 10 0D 0D 5B 5F 5B 00 26 00 00 03 00 02 00 00 52 00 01 00 05 34 0B 0D 13 5C 63 52 00 32 07 00 00 00 04 03 02 4A 01 00 1C 00 34 0B 0D 23 5C 63 52 00 32 07 00 00 00 04 00 02 63 01 00 1C 00 3E 0B 0E 0C 5C 63 5C 00 24 17 03 00 00 01 00 01 50 00 01 00 0A 34 0B 0D 23 5C 63 52 00 32 07 00 00 00 04 00 02 63 01 00 1C 00 57 11 3C 62 32 32 32 32 01 07 00 1B 37 30 00 01 00 01 18 48 65 72 61 20 20 20 20 20 20 00`
MonoMode1=0
ModulationWheelRange1=99
.
.
.

probonopd, 2023-08-18:
Sooner or later the point is approaching where we need to get our hands on a TX802 to learn how it handles a few things ;-)

diyelectromusic, 2023-08-18:
Well if someone wants to buy me a TX7, TX802 and TX816, I promise to do a full and thorough comparison :)

probonopd, 2023-08-20:
fwiw, the emulator [MAME](https://www.mamedev.org/) (known for emulating arcade games) can also emulate a few synths, including the TX81Z. Te way I understand it, it emulates the chips, and runs the original firmware. Maybe it can be helpful for experimenting.

https://www.youtube.com/watch?v=Dj-n3HQNch4

diyelectromusic, 2023-09-19:
> fwiw, the emulator [MAME](https://www.mamedev.org/) (known for emulating arcade games) can also emulate a few synths

I keep looking in on this from time to time, but I've never really found any concrete information on how to actually do it...  Personally I'd like a CZ equivalent of MiniDexed but as far as I can see, no-one has gone down that route.

Kevin

Banana71, 2023-12-04:
Personally, I don't think much of an automatic MIDI dump. With an automatic dump, I fear that voice data in Dexed (PC-VST) will be overwritten if you load another performance after a dump.
How about a MIDI menu with the MIDI Dump submenu from which you can specifically select a voice (between TG1-TG8) for the MIDI dump. Dexed could then receive the voice data and process it further.
Or would it be possible to send the 8 voices (+24 empty voices) from a performance as a sound bank in the sysex to e.g. MIDI-OX?

probonopd, 2025-04-20:
`MIDIAutoVoiceDumpOnPC` has been implemented. So can this be closed @Banana71?

(Feel free to open other issues for other topics, such as saving voice sysex.)