probonopd/MiniDexed #547: USB Gadget Support

diyelectromusic, 2023-09-30:
I'm just parking this here for now as a reminder to take a look at some point: https://github.com/rsta2/circle/discussions/387

This would be nice to include once the bugs have been shaken out.

Kevin

probonopd, 2023-10-28:
Interesting! Looks like `CUSBMIDIGadget` landed in Circle [release 45.3.1](https://github.com/rsta2/circle/releases/tag/Step45.3.1).

probonopd, 2023-10-28:
Let's see the answers to https://github.com/rsta2/circle/discussions/387#discussioncomment-7412396

diyelectromusic, 2023-10-28:
Yep.  I'll take a look when I've stopped messing around with 16 tone generators! :)

I don't think its a big change - all the hard work is done in circle.

Kevin

diyelectromusic, 2023-11-05:
BTW @probonopd would you like to see if MiniDexed could request a VID/PID for use as a USB Device?  I can make things up for experimenting, but for proper builds there ought to be a proper code - the alternative is that users have to find their own, but they can't be set without recompilation at present so that would really limit their use!

Options seem to be:
- But one from USB foundation.  Cost ~$6000... :)
- Via Raspberry Pi, but this looks more for people building actual products on a Pi: https://github.com/raspberrypi/usb-pid
- Via pid.codes which looks more for open source projects: https://pid.codes/howto/
- OpenMoko appear to be allowing their VID for use with open source projects too: https://wiki.openmoko.org/wiki/USB_Product_IDs

I think we only need one VID/PID as all our builds are for a class compliant MIDI USB device, so driver wise, it should all be fine at the OS end...

I could do it of course, but I thought a request from your own account would be more appropriate?

Kevin

probonopd, 2023-11-05:
Wouldn't it be sufficient to have a USB VID/PID for Circle?

diyelectromusic, 2023-11-05:
That's a question for Rene, really, but as he had the choice already and has explicitly decided not to...

In future I'd imagine Circle may support other USB device modes, not just MIDI, so I think the right place is in the projects that are using circle rather than the core code itself.

Kevin

probonopd, 2023-11-05:
Applied at https://github.com/pidcodes/pidcodes.github.com/pull/895. Thanks @diyelectromusic

diyelectromusic, 2023-11-05:
Update:

I now have USB Gadget mode working on the following that I've been able to test:
* Raspberry Pi A (kernel.img)
* Raspberry Pi 3A+ (kernel8.img)
* Raspberry Pi 4B (kernel8-rpi4.img)
* Raspberry Pi Zero (kernel.img)
* Raspberry Pi Zero W (kernel.img)
* Raspberry Pi Zero 2 (kernel8.img)

This probably means it will work on the Pi 1 A+ as that is supposed to support Gadget mode too.  I have no idea about the compute modules, although some of them apparently support it too.  I don't think there was a variant of a Pi 2 that supported it.

IMPORTANT: This is NOT ready for use yet for a number of reasons, but mostly that the USB ID question needs resolving first.  Also, ideally it would have more robust checking around the Pi board and if it supports USB Gadget mode or not, but I'm still chewing over how best to do that.

But you can follow along with the latest here: https://github.com/diyelectromusic/MiniDexed/tree/USBGadgetDevice

Notes on powering: In USB Gadget mode, the Pi should be powered over the USB bus from the host.  But there may be cases when the host can't supply enough power to the Pi.  However, a Pi MUST NOT be powered from a USB host and a separate power supply at the same time!  People have been finding inventive ways to get around this, but I'm not getting into that here!  There is some discussion here: https://forums.raspberrypi.com/viewtopic.php?t=352564

For the above boards, this means the following connections for a bus-powered system:
* Pi A or 3A+ - use a USB A to USB A lead to connect the Pi 3A+ to a USB hub/host.  The micro USB socket is NOT to be used.
* Pi 4B - use the USB C socket with a USB C to USB C or USB C to USB A lead.  The USB A sockets are NOT to be used.
* Pi Zero, Zero W, Zero 2 - use a micro USB to USB A lead on the "USB" port (not the "PWR IN" port).

In all cases, the Pi can be in only one of either Gadget or Host mode.  This is a configuration setting USBGadget=(0|1) in minidexed.ini (default=0).  It can't be changed "on the fly" - it has to be set at power up.

And it is IMPORTANT that the Pi is connected correctly for the right USB mode.

In short: this will soon be possible. But I don't plan to include it in minidexed.ini by default **as I consider it an Advanced option**, due to the issues around supplying power.  I will however document it on the wiki once it is all ready for the bold and adventurous among us :)

Massive thanks as always to Rene as always for doing all the hard work in his Circle environment in the first place.

Kevin

kaicherry, 2023-11-08:
@diyelectromusic  - Kevin, I built and ran this yesterday, after deducing how to get USB Ids and such in there, and I have to say, just "plugging my box into my Mac" and "there it is" was a *real treat* and I hope...something is worked out soon. Thanks for adding this, man...it's great. Much better "FT/OOTB User Experience" :)
-K

diyelectromusic, 2023-11-08:
> @diyelectromusic - Kevin, I built and ran this yesterday, after deducing how to get USB Ids and such in there,

Haha - that is quick off the mark!  Yes, I haven't included the USB ID bit yet as we're waiting for one to be allocated to MiniDexed.  But I agree I think this will be a really fun one.  I have some of those "USB Stem" add-ons for a Pi Zero that I want to try, but I think this will have to remain an "advanced option" for the time being due to the issues around having to understand how the powering of the thing works (not a major issue for the Zero, but more of an issue for the later Pis).

I'll write up a special USB Gadget page on the Wiki once it is all ready though.

And I've done the easy bit.  All the hard work was done by Rene in Circle.

Kevin

kaicherry, 2023-11-09:
Ah I *see*...yes, if that's a path (specific vendor/device ID) that is *wonderful*...what I did as a "cheat" but 'honest one' was use the Adafruit vendor ID. I (very briefly) scanned the source and believe I told myself Circle is using TinyUSB, so "that's fair" :)

I *did* have another question however...is there a way to "re-map" the rotary to match the (I know..I know...) mt32-pi scheme...without changing the source code? I wanted to build a "brain" for my son that would work with both...tho I personally think having a "desktop" DX7x8 "rack module" on your desk is *cooler*,,but 'kids today with the samplers and the beat boxes" don'y know what an A, D, S, nor R is so... XD

At any rate, following that effort "over there with the USBz" closely :)

-K

diyelectromusic, 2023-11-09:
Last time I looked MT-32 Pi only supported volume changes with the rotary encoder - it hadn't implemented the Roland MT32 rotary encoder use as I recall?

So, no, not really.  The encoder is pretty crucial to navigating the menus.... so my advice if you wanted to build an extra thing that could support either would be to drive them both over MIDI using a common device based on a microcontroller driving an independent UI.  That isn't easy to do and still use the display however, so the Pi part of them both would probably have to be largely "headless" and then relying on whatever CC and program change messages are exposed over MIDI.

But at that point you're basically reimplementing any of the available MIDI controllers, so you might just be better off getting one with programmable control schemes that can be called up from the controller itself.  My Arturia Minilab allows that for example, but you definitely end up with a "lowest common denominator" kind of interface :)

Kevin

kaicherry, 2023-11-09:
"Details matter"  - I didn't mean simultaneously :D 
What I had in mind was more like a physical toggle switch on a GPIO. In my (likely, warped) mind, I was thinking of some sort of "hook" that would switch config.txt; if the synth is powered up and the is ON, run/stat/hook the minidexxed "kernel"/run mini, OFF, mt32. I may have time to mess with it over the weekend. Should I fork this repo, or the other one tho :)
-K

probonopd, 2023-11-20:
...and we now have a USB ID 💯: 
Vendor `0x1209`, device `0xF043`

https://github.com/pidcodes/pidcodes.github.com/blob/master/1209/F043/index.md

closed 2023-11-27 by diyelectromusic