probonopd/MiniDexed #549: Automatically see the parameter and its value on the screen when we edit or control it externally via SysEx.

RiNoize, 2023-10-14:
I am developing an arduino based midi controller, a qwerty keyboard and 8 potentiometers.
The idea is to control all the internal functions and parameters of MiniDexed via SysEx messages.
I am copying the SysEx messages from the Dexed, I put them in my Arduino controller and from there I send the message, so far I see that they work in the Minidexed, the issue is that to see the value of the parameter that I am acting on I have to enter edit voice mode from MiniDexed, go to the parameter and there I can see what value I am giving it from the potentiometer of the Arduino controller.
Would there be any possibility of automatically seeing the parameter and value of the parameter that we are acting on without having to enter edit mode?
For example: I am in voice mode, TG1, and I have the name of Patch 1 = PIANO 3 on the screen
But when you act on any knob of the Arduino controller to control a certain parameter of the MiniDexed, the parameter with the value that we are changing appears on the screen, then after 3 seconds of stopping acting on the parameter, it automatically returns to the name of the patch or voice .
Is it possible to do that?
My programming knowledge is limited, I am programming in Arduino using ChatGPT.
Thank you in advance for such a good and fun project.

probonopd, 2023-10-28:
Excellent suggestion. I like it.
Not sure how easy it is to implement though. Maybe @diyelectromusic or @lexonic could have a look?

probonopd, 2025-04-20:
In other words:

When we receive a MIDI CC, simulate the user navigating to the area of the menu UI (for the first of the TGs that match the MIDI channel on which the CC was received) that deals with the corresponding parameter (e.g., Detune if the CC for Detune is received). Then, after 2 seconds, go back to where in the menu the user was in the menu structure before the MIDI CC had arrived.