probonopd/MiniDexed #562: Modulation Range Error

Banana71, 2023-11-17:
If any modulation range is changed, the modulation can no longer be heard; the modulation only works again after blindly saving and reloading the performance.
You can easily try it out with the performance 000011_SeventyOne, for example. Simply turn the modulation wheel all the way up until the sound has a lot of vibrato, then change the modulation range for the modulation wheel from 99 to 98 in the settings of TG1 and TG2 and you can no longer hear vibrato.
The problem occurs with all types of modulation. (ModWhell, Breath Controller, Foot Controller, Aftertouch)

probonopd, 2023-11-17:
Thanks for reporting this issue @Banana71. Do you happen to know whether this works correctly in Dexed on the PC?

Banana71, 2023-11-17:
In Dexed everything works as it should. Range changes are applied immediately, even when tones are pressed and held.

probonopd, 2023-11-17:
@dcoredump can you reproduce this on MicroDexed?

dcoredump, 2023-11-19:
I have to check this, give me some time. I think I know what's going wrong...

Thanks @Banana71 for deep testing and finding problems I never thought they may exist 😁

dcoredump, 2023-11-20:
I created a PR. Hope this fixes the problems. Cannot test - no keyboard right here.

Banana71, 2023-11-20:
Hi Holger @dcoredump , 
Unfortunately the error is still there

probonopd, 2023-11-20:
@Banana71 please try this build:
[MiniDexed_2023-11-20-8d1e592](https://github.com/probonopd/MiniDexed/suites/18351374688/artifacts/1062168230)

It is built using the fixed version from the posting above.

Banana71, 2023-11-20:
The error has changed.
I set the Mod.Wheel to MAX, the sound has full vibrato. After adjusting the Mod.Wheel range from e.g. 99 to 80, the vibrato remains at MAX and the vibrato stops completely when you move the Mod.Wheel.
In other words: you can adjust the range but the modulation remains at the old value. After moving the Mod.Wheel the modulation goes off and can no longer be restored.

dcoredump, 2023-11-21:
Thanks for testing!

Ok, I need to get my MIDI keyboard from the rehearsal room to my lab at home for deeper inspection. I hope I can try to dive in deeper at next weekend.

dcoredump, 2023-12-23:
Hi Peter,

now I got some time for testing this with MicroDexed (see this [video](https://youtu.be/rnkaJv1xKww?feature=shared)). On MicroDexed it works as expected. So the problem may be located inside MiniDexed?

Regards and merry xmas, Holger

Banana71, 2023-12-24:
Thank you Holger for your effort. 

Merry Christmas to you all 🎄
Peter

Banana71, 2025-05-03:
Good things come to those who wait. For whatever reason, probably due to the Glissando Effect Fix https://github.com/probonopd/MiniDexed/issues/889 , the modulation range parameterization now works. 
Thanks

Or does anyone have an explanation for why this suddenly works?

probonopd, 2025-05-03:
Glad to hear it works now! 👍 

I don't have a better explanation.

closed 2025-05-03 by probonopd

dcoredump, 2025-05-15:
> Or does anyone have an explanation for why this suddenly works?

Maybe I fixed it and forgot to tell this for upstream integration...

soyersoyer, 2025-05-15:
Maybe this one? https://codeberg.org/dcoredump/Synth_Dexed/commit/b9fd3010f28ba248049315dbd81c887bdc3f3a38