probonopd/MiniDexed #563: Reverb time depends on the sample rate

Banana71, 2023-11-19:
I wanted to improve the latency of the miniDexed and experimented with the SampleRate and ChunkSize in the miniDexed.ini. I noticed that the ReverbTime behaves anti-proportionally to the SampleRate.

Is it possible to change the programming so that the reverb time is independent of the sample rate or should we just document the behavior?

Why all this: I noticed an audible latency between the miniDexed and the Roland RD-300 for sounds with a short attack time. With the following values, the latency can no longer be determined and even a Raspberry Pi 2 runs stable:
`SampleRate=48000` (default)
`ChunkSize=128` (default 256)

With a sample rate of 96000, the latency is measurably better but the reverb time is halved.

probonopd, 2023-11-19:
Good observation! In other words, you are saying that we need to adjut the reverb time like this?

```
AdjustedReverbTime = ReverbTime / (DefaultSampleRate / SampleRate)
```

probonopd, 2023-12-09:
@Banana71 can you confirm the above? Then we could go about implementing it.

Banana71, 2023-12-09:
```
AdjustedReverbTime = DesiredReverbTime * correctionFactor

correctionFactor = SampleRate / DefaultSampleRate
```

The formula describes the correct solution approach.

But the code for the reverb effect already works with a sample rate. Is the code in the effect incorrect? Are the variables not assigned correctly? What sample rate is mentioned in the code for the effect?

search for Samplerate:
[image]

Banana71, 2025-04-20:
Thanks alot @probonopd.  🥇

Banana71, 2025-04-20:
I was getting too excited.

I've now reloaded the version MiniDexed_2025-04-18-b15ffad. At sample rates < 48000, the reverb time gets shorter. Now the behavior is exactly the opposite.

At a sample rate of 96000, miniDexed crashes.

When I first tested it, I was probably just listening to see if the reverb time increased at 44,100 Hz. The difference between 44,100 and 48,000 is small. I hadn't noticed that then.