probonopd/MiniDexed #572: Last note aftertouch

probonopd, 2023-12-01:
We should implement "last note aftertouch", in which monophonic aftertouch (not polyphonic aftertouch) capable MIDI controllers can control the last pressed note only. It should be a configuration setting whether monophonic aftertouch should affect all notes (as it is now) or only the last note.

Idea credits:
youtube.com/watch?v=eg6X9fQ-U4s

probonopd, 2023-12-02:
@dcoredump what do you think, would this be a good fit for Synth_Dexed?

dcoredump, 2023-12-04:
If you want to control individual notes with AT, you definitely have to make some changes in the depths of Dexed. A "last-note-AT" would virtually fall off if you implement poly-AT. In this respect, I think it would be better to implement polyphonic AT first.

I find this task very exciting, but I want to fix the various errors in the current code first - and I don't even have the time for that at the moment. So if someone should find someone - I am happy to be available as a source of information.